OpenTabletDriver/OpenTabletDriver #2155: A15 pro,  Config file you gave me now detects tablet, but when pen is hovered over tablet cursor goes crazy up and down

DrunkenMasterTim, 2022-04-02:
# Support Request

## What I've already tried

## Diagnostic Information

[New Text Document.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/8402865/New.Text.Document.txt)

jamesbt365, 2022-04-02:
No point making **another** issue over the same combination of problems, i haven't gotten around to checking the configuration yet as its not complete.

closed 2022-04-02 by jamesbt365

DrunkenMasterTim, 2022-04-02:
> <OpenTabletDriver/OpenTabletDriver/issue/2155/issue_event/6357095305@
> github.com>

How am I supposed to know, I've never had an issue with the software until
now, and why would you give me an unfinished configuration file? Not trying
to be rude

jamesbt365, 2022-04-02:
Can i finish a configuration file without knowing if it works at all in the first place? No i can't, how am i meant to predict if your variant of the tablet is going to work 100% as the previous one, no i can't, i can't predict and fix problems before they arise, wait patiently i'm busy.

DrunkenMasterTim, 2022-04-02:
Thank you for the clarifications