OpenTabletDriver/OpenTabletDriver #2159: OpenTabletDriver Error: ExtendedSocketException Permission denied.

cindrmon, 2022-04-03:
## Description
I was trying to open OpenTabletDriver on Fedora 35, and this always came up. It doesn't detect my Huion H420 tablet like it used to when I first installed it anymore. Is this something that had already been fixed in v0.6.0.3? Because, I can't install the latest version due to #1986.

## System Information:
| Name             | Value |
| ---------------- | ----- |
| Operating System | Fedora 35         |
| OpenTabletDriver Version | v0.6.0.2 |
| Tablet       | Huion H420            |

jamesbt365, 2022-04-03:
Stacktrace? Sounds like you tried to run otd as root then as regular user. reboot.

cindrmon, 2022-04-03:
> Stacktrace? Sounds like you tried to run otd as root then as regular user. reboot.

Oh i forgot to send the stack trace, i apologise.

[image]

And i attempted to restart my pc, and it still shows the same error as before even after just logging in.

jamesbt365, 2022-04-03:
otd is infact running as a different user, which is why it cannot connect.

cindrmon, 2022-04-03:
how is that possible? i only have one user in fedora 35.

jamesbt365, 2022-04-03:
you may have told the service to autostart under a different user

cindrmon, 2022-04-03:
which service in particular?

jamesbt365, 2022-04-03:
`opentabletdriver`

cindrmon, 2022-04-03:
how do i find the status of the `opentabletdriver` service? can i do that with `systemctl`?

jamesbt365, 2022-04-03:
as it is a systemd user service, yes.

jamesbt365, 2022-04-03:
if you are 100% sure its not running under a different user remove `/tmp CoreFxPipe_OpenTabletDriver.Daemon`

cindrmon, 2022-04-03:
this is the output when i want to check the status for `opentabletdriver`:
```sh
-> % systemctl status opentabletdriver
Unit opentabletdriver.service could not be found.
cinder@nexvsalphaxiii [20:15:23] [~] 
-> % systemctl status opentabletdriver@cinder
Unit opentabletdriver@cinder.service could not be found.
cinder@nexvsalphaxiii [20:15:27] [~] 
-> % 
```

jamesbt365, 2022-04-03:
its a user service.

cindrmon, 2022-04-03:
> ```shell
> -> % systemctl status opentabletdriver
> Unit opentabletdriver.service could not be found.
> cinder@nexvsalphaxiii [20:15:23] [~] 
> -> % systemctl status opentabletdriver@cinder
> Unit opentabletdriver@cinder.service could not be found.
> cinder@nexvsalphaxiii [20:15:27] [~] 
> -> % 
> ```

yes, i know what you said. but as you saw here, i even tried checking it as a user service, and it couldn't be found as well.

jamesbt365, 2022-04-03:
...

`systemctl --user status opentabletdriver`

cindrmon, 2022-04-03:
ahh i see, apologies. this is the output:

```sh
● opentabletdriver.service - OpenTabletDriver Daemon
     Loaded: loaded (/usr/lib/systemd/user/opentabletdriver.service; enabled; vendor preset: enabled)
     Active: activating (auto-restart) (Result: core-dump) since Sun 2022-04-03 20:19:55 PST; 25s ago
    Process: 25617 ExecStart=/usr/share/OpenTabletDriver/OpenTabletDriver.Daemon -c /usr/share/OpenTabletDriver/Configurations (code=du>
   Main PID: 25617 (code=dumped, signal=ABRT)
        CPU: 743ms

Apr 03 20:19:55 nexvsalphaxiii systemd[2867]: opentabletdriver.service: Main process exited, code=dumped, status=6/ABRT
Apr 03 20:19:55 nexvsalphaxiii systemd[2867]: opentabletdriver.service: Failed with result 'core-dump'.
```

jamesbt365, 2022-04-03:
well disable the service under probably `root`, delete `/tmp/CoreFxPipe_OpenTabletDriver.Daemon` then start it under your user. Probably when you installed it and you went to start the service you did it using root (either manually with su or sudo/doas)

closed 2022-04-03 by jamesbt365

cindrmon, 2022-04-03:
ohhh i see thanks