OpenTabletDriver/OpenTabletDriver #2160: Add support for XP-Pen Deco Fun S

Rondosu, 2022-04-04:
# Tablet Information

| Name            | Value |
| --------------- | ----- |
| Manufacturer    |XP-PEN
| Product Name    |Deco Fun S (CT640)
| Width           |6.3"
| Height          |4"
| Pressure Levels |8192
| LPI             |5080

James already replied to me saying that they have the CT640.json
I am still confused on what to do so I asked him more question but he didn't reply
so I decided to post it again but with the tablet information this time
sorry if I cause the disturbance >~<
I'm just not a big fan of the official driver for the deco fun

jamesbt365, 2022-04-04:
This tablet is already supported, give me more explanation of why you opened another issue.

Rondosu, 2022-04-04:
but I downloaded the current release but it not detecting my tablet

jamesbt365, 2022-04-04:
Help > export diagnostic, save that file and send it here as a `.txt`

Rondosu, 2022-04-04:
how do I do that?
sorry I am new to github

jamesbt365, 2022-04-04:
Help > export diagnostic is in opentabletdriver, and then you drag and drop the file it saves.

Rondosu, 2022-04-04:
it says we don't support that file type

jamesbt365, 2022-04-04:
Thats why i said save it as a txt

Rondosu, 2022-04-04:
oh ok
sorry for misunderstandin
[help diagnostic.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/8409709/help.diagnostic.txt)

jamesbt365, 2022-04-04:
- Download [this](https://raw.githubusercontent.com/OpenTabletDriver/OpenTabletDriver/configs/OpenTabletDriver.Configurations/Configurations/XP-Pen/CT640.json) file
- make a `Configurations` folder be the executable of otd
-  move the file you downloaded there
- Tablet > detect tablet

closed 2022-04-04 by jamesbt365

Rondosu, 2022-04-04:
When I click on the download file link
it bring me to a page with bunch of texts
do I just copy them all and put them into a text file or?

jamesbt365, 2022-04-04:
Right click the page or control + s, you are downloading it.

Rondosu, 2022-04-04:
btw what do you mean by the "make a configurations folder be able to executable of otd"?
and move the files to where?

Rondosu, 2022-04-04:
sorry, not really good at these sorts of things

jamesbt365, 2022-04-04:
You make a folder, called `Configurations`, next to the executables (OpenTabletDriver.UX.Wpf.exe and OpenTabletDriver.Daemon.exe), then you move that file you downloaded (CT640.json) into the `Configurations` folder.