OpenTabletDriver/OpenTabletDriver #2163: Support Request: Tablet detected, but not moving cursor

WaveLinkdev, 2022-04-06:
# Support Request
Tablet detected, but not moving cursor.
OS: Arch linux with zen kernel
Tablet: Wacom CTE-640
I also have 2 monitors connected if it makes a difference.

The tablet shows up in otd, but using the pen on the tablet doesn't do anything. When moving the pen over the tablet, otd shows raw tablet data as it should in the tablet debugger, so I believe that is not specifically a problem of reading the tablet data.

## What I've already tried
I have tried every tip in the linux support forms
Reinstalling the `opentabletdriver` package from the AUR as well as the `opentabletdriver-git` package.
Disabling the `wacom` module

## Diagnostics
[diagnostics.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/8429018/diagnostics.txt)
ostic Information

Thanks for the help! 😄

Sublimelime, 2022-04-06:
Hey there,

You have Wacom's drivers installed. OTD puts a console message warning under the console tab when it detects this. As they're incompatible (OTD serves as a replacement) you'll need to blacklist the wacom module:

```echo "blacklist wacom" | sudo tee -a /etc/modprobe.d/blacklist.conf
sudo rmmod wacom```

After this, reboot. Should function after that.

jamesbt365, 2022-04-06:
Download [this](https://github.com/OpenTabletDriver/OpenTabletDriver/files/8429121/CTE-640.json.txt) file then remove the `.txt` at the end leaving it just as `CTE-640.json`, create the directory `/usr/share/OpenTabletDriver/Configurations` move the file there.

WaveLinkdev, 2022-04-06:
Hey! That seems to have done the job. After restarting the daemon the cursor now moves, thanks for the help!

closed 2022-04-06 by WaveLinkdev