OpenTabletDriver/OpenTabletDriver #2166: GUI does not open

Saxorocheck, 2022-04-06:
I apologize for my poor English, as I don't speak it perfectly. The graphical interface of the driver stopped opening after an error in the minecraft game, I do not know if this is somehow related. When I try to open the GUI, it opens, stops for a few seconds and closes. Help(

jamesbt365, 2022-04-06:
Fill out the template, i have no idea what your system. Not even the Operating system. Therefore i'm assuming its windows.

- First make sure that there is no Configurations folder if you are using any version higher than 0.5.3.3.
- Delete or move %localappdata%\OpenTabletDriver\settings.json
- Delete or move %localappdata%\OpenTabletDriver\Plugins

closed 2022-04-06 by jamesbt365

Saxorocheck, 2022-04-06:
> Заполните шаблон, я понятия не имею какая у вас система. Даже не Операционная система. Поэтому я предполагаю его окна.
> 
> * Сначала убедитесь, что папки Configurations нет, если вы используете версию выше 0.5.3.3.
> * Удалить или переместить %localappdata%\OpenTabletDriver\settings.json
> * Удалить или переместить %localappdata%\OpenTabletDriver\Plugins

Thank you, you helped me a lot! Thanks again.