OpenTabletDriver/OpenTabletDriver #2169: Support Request: Huion HS610 no being detected

soverte, 2022-04-08:
# Support Request

## What I've already tried

## Diagnostic Information

Sublimelime, 2022-04-08:
Please fill out the template.

soverte, 2022-04-08:
> Please fill out the template.

sorry is miss click it, also dont speak english so im triyng

jamesbt365, 2022-04-08:
Right click the windows logo > device manager > view > device by container > find your tablet > open the container > right click the entries inside > uninstall each of them, once you have done all of them then replug your tablet.

soverte, 2022-04-08:
> Right click the windows logo > device manager > view > device by container > find your tablet > open the container > right click the entries inside > uninstall each of them, once you have done all of them then replug your tablet.

soverte, 2022-04-08:
> Right click the windows logo > device manager > view > device by container > find your tablet > open the container > right click the entries inside > uninstall each of them, once you have done all of them then replug your tablet.

thanks dude it works.

closed 2022-04-08 by soverte