OpenTabletDriver/OpenTabletDriver #2172: Turning on artist mode makes pen buttons no longer work

Joanthenoob, 2022-04-09:
## Description

like the title says switching to artist mode makes the pen buttons stop working. They do however work in absolute mode.
I tried loading artist mode through a preset and still the same issue.

Reproducible: everytime 

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         | Arch Linux 
| OpenTabletDriver Version | v0.6.0.3+1ac31a7
| Tablet                   | Huion H950p

jamesbt365, 2022-04-09:
This doesn't matter the output mode, one bindings from an output mode cannot be used on another. This leads to where you can't use regular mouse buttons for modes like windows ink, vmulti mode and modes like artist mode in this case. This would be useful to use in different output modes.

gonX, 2022-04-09:
This smells a little bit like #1767 

We've considered the option of allowing mouse buttons to be used in Artist Mode, but I was personally against it since we'd be sending the appropriate pen buttons instead, which we have no clear way of indicating to the user.

I'm in favor of having known-good mapping translations done on an output mode switch, since most people likely won't care what their bindings change to as long as it works. It's an important distinction since some applications have the potential to actually treat pen tip and left click differently on Linux.

sajmon170, 2022-06-03:
I also have this issue, and for me it makes it impossible to do any serious work. My workflow required the pen buttons to be remapped to right click and middle click respectively. Now I can't drag my canvas and I can't use the color/brush selector.

jamesbt365, 2022-06-03:
for now you can use script runner and xdotool but if you want it natively you'll have to wait.

sajmon170, 2022-06-03:
Do you know how to set this up on Wayland? Xdotool works only on X11, and I can't switch to it since the latest GNOME update made it lag a lot. It seems that Xorg's getting phased out by the dev team.

gonX, 2022-06-04:
If your WM/DE is any good, the usual pen buttons should be treated as equivalent mouse buttons. Last time I tried getting sending actual mouse buttons events, libinput just discarded it, so if the pen button bindings (picture below) aren't working for you, we'd have to look deeper into libinput.

[image]

sajmon170, 2022-06-04:
@gonX I use GNOME 42 with a Huion Q11K V2 drawing tablet and my pen button bindings are empty. I can't set them to match your screenshot.
[image]

X9VoiD, 2022-06-04:
Click on the `...` button

sajmon170, 2022-06-04:
Thank you! I managed to set it up and it works like a charm.

X9VoiD, 2022-06-04:
@Joanthenoob refer to the messages above for the fix. Will close now as resolved

darkingmimo, 2025-08-06:
the problem still persists. I mapped my pen binding, went into artist mode. Nothing. They only work for absolute mode.

closed 2025-12-06 by gonX