OpenTabletDriver/OpenTabletDriver #2180: [DllNotFoundException:Error] Unable to load shared library 'libXrandr' or one of its dependencies

sigaloid, 2022-04-11:
## Description
I am getting this error: 
[image]

I get this while on X11 in both i3wm and Gnome X11. Notably not on Gnome Wayland probably because of Xwayland. 

I have installed all of the recommended packages with commands like `sudo dnf install libX11 libX11-devel libX11-common libX11-xcb libXrandr libXrandr-devel`, `sudo dnf install libX11-* libXr* libevdev*`, `sudo dnf install libX11-devel libXrandr-devel libevdev-devel`, etc. I believe I have installed every fedora equivalent of the requirements.

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         |Fedora 35 
| OpenTabletDriver Version |v0.6.0.3
| Tablet                   |Gaomon S620

[diagnostics.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/8467460/diagnostics.txt)

jamesbt365, 2022-04-11:
Try making an rpm based off this pr:
https://github.com/OpenTabletDriver/OpenTabletDriver.Packaging/pull/16

sigaloid, 2022-04-11:
I thought I generated an RPM but then it deleted it: 
[image]

Can you explain how to generate it?

ken-soares, 2022-04-21:
Any update on how to fix the issue?

jamesbt365, 2022-04-21:
its probably solved upstream, though nobody has generated an rpm off that pr and used it to see.

PR #2273 (2022-05-30): Explicitly use libX11.so.6 and libXRandr.so.2
This fixes issues that pops up on recent distros like Fedora.

Resolves #2180
closed 2022-05-30 by InfinityGhost