OpenTabletDriver/OpenTabletDriver #2188: hello, i would need help setting up the Gaomon M10k pro

squa81, 2022-04-14:
# Support Request

## What I've already tried

## Diagnostic Information

jamesbt365, 2022-04-14:
one, you have forgotten to fill out the template, two this is being dealt with in discord.

squa81, 2022-04-14:
i am currently udating the template to fill out the requirement, i didn't have the time yet, and this has been going on discord for 2 days now

jamesbt365, 2022-04-14:
[image]
we have even asked for information that you haven't provided ¯\_(ツ)_/¯. And yes, jank tablets require alot of time.

squa81, 2022-04-14:
i can't be on two places at the same time

jamesbt365, 2022-04-14:
This was opened after, instructions were given, do not just paste diagnostics in either. Attach them as a file. On top of this saying "i tried getting help on discord" doesn't say what exactly you got help with, "i got help" doesn't help us

squa81, 2022-04-14:
this has been updated to solved on discord

jamesbt365, 2022-04-14:
[image]
if its solved this button does infact exist and does work.

closed 2022-04-14 by jamesbt365