OpenTabletDriver/OpenTabletDriver #2192: Support Request: Not detecting the XP PEN Artist 15.6

Julian-Heidenreich, 2022-04-16:
# Support Request

The Tablet is not detected. I use an XP PEN Artist 15.6 and assumed that the entry in the supported list for the 15.6 pro includes mine too.

## What I've already tried

Cleaned up previous drivers.
Restarted my PC.
Reinstalled the Desktop Runtime.
Tryed diffrent USB ports (2.0, 3.0, 3.1)
All from the Official website

## Diagnostic Information

[otddia.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/8500040/otddia.txt)
(GitHub doesn't accept json formatted file for upload)

jamesbt365, 2022-04-16:
Isn't supported in master, but is in configs, let me adjust the configuration real quick.

jamesbt365, 2022-04-16:
in a minute or 2, a windows action build should be available that will hopefully detect the tablet, however i'm not sure that aux buttons will work yet and have no clue if it will even work correctly with this configuration. Download it from [here](https://github.com/jamesbt365/OpenTabletDriver/actions/runs/2177013101), extract it to its own folder and it should pick up the tablet, if it does go into `tablet > tablet debugger` and get the max x and y coordinates at the bottom right of the tablet, (the visualizer may be incorrect so ignore it). Also check if the buttons work if they are bound.

Julian-Heidenreich, 2022-04-16:
Ok the Tablet was detected properly and the buttons on the tablet are working too. But I am not sure if the debugger is working right. I open it and nothing is shown. (No device selected and no raw input)

jamesbt365, 2022-04-16:
try this one. https://github.com/jamesbt365/OpenTabletDriver/actions/runs/2177145944

Julian-Heidenreich, 2022-04-16:
The debugger works fine now.
The Aux buttons are getting registered in the debugger but I cant bind them in the Auxiliary Settings
Additionally to that the sensitivity of the pen isn't working anymore. That as well is working fine in the debugger but if I open photoshop f.e. it isn't working properly

TY for your help btw :)

jamesbt365, 2022-04-16:
I still need the max x and y in the debugger (located at the bottom right of the tablet, i need these to be able to assign the correct size to the tablet if it is incorrect, as well as making sure the configuration is accurate), buttons should be in one of the tabs, should bind correctly. Pressure support you have to use a plugin for. Could you verify everything words accordingly?

Julian-Heidenreich, 2022-04-16:
The max I got was X 34411 and Y 19460

jamesbt365, 2022-04-16:
Check the tabs for the tablet buttons again, they should be there to bind.

Julian-Heidenreich, 2022-04-16:
Ok I deserve punishment for not hitting apply after setting the bindings...

jamesbt365, 2022-04-16:
lmao, all good. Everything work then?

Julian-Heidenreich, 2022-04-16:
As I see yes. I still try to work out the sensitivity thing but I should get that on my own.
Just a question after that. Will this now go into the master branch after that? And until then I use this version you provided me? or how does this work now?

jamesbt365, 2022-04-16:
A release will be coming out soon, either today or in the next few days. Once void merges this configuration into configs branch it will be pulled into master when the next version is released so you won't need to use the build anymore.

and for your other question:
https://github.com/X9VoiD/VoiDPlugins/wiki/WindowsInk

Julian-Heidenreich, 2022-04-16:
Ah ok it is this Plugin ty :D

Did I see this right you just added the identifier for the 15.6 2 days ago? What an coincidence xD

jamesbt365, 2022-04-16:
_yep_

Julian-Heidenreich, 2022-04-16:
Ok thank you so much for the help and have nice eastern ;D

PR #2193 (2022-04-16): Add identifier for XP-Pen Artist 15.6
fixes #2192
closed 2022-04-16 by Julian-Heidenreich