OpenTabletDriver/OpenTabletDriver #2198: Add support for Gaomon M6 Tablet

Stehsaer, 2022-04-18:
# Tablet Information

| Name            | Value |
| --------------- | ----- |
| Manufacturer | Gaomon  |
| Product Name | Gaomon M6  |
| Width | 254mm |
| Height | 158.8mm |
| Pressure Levels | 8192 |
| LPI | 5080 |

## Diagnostic Information

[diag.json.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/8505575/diag.json.txt)

jamesbt365, 2022-04-18:
Tablet > device string reader at the top of otd.

- 9580 top box
- 100 middle
- dump all

save that file and send it here

Stehsaer, 2022-04-18:
[dump.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/8505695/dump.txt)

jamesbt365, 2022-04-18:
- Download [this](https://github.com/OpenTabletDriver/OpenTabletDriver/files/8505738/M6.json.txt) file
- Make a `Configurations` folder next to the executables of otd.
- Move this file here
- tablet > detect tablet.

Stehsaer, 2022-04-18:
[image]
tablet detected

jamesbt365, 2022-04-18:
Go into tablet > tablet debugger, get the max x and y coordinates at the bottom right of the tablet. (ignore the visualizer)

while you are there, can you check if the pen buttons work properly in the debugger (there should be 2)

Can you get the max pressure as well by pressing the pen down

Does this tablet have tilt?

How many tablet buttons does this tablet have (not counting pen)?

Stehsaer, 2022-04-18:
- Position: 
[image]
- Both pen buttons work well as expected
- Get a max pressure of 8191 when pushed hard against the surface
- The purchasing page of this tablet claims that it has tilt and I can see a significant change in the last 2 values of raw tablet data in the debugger when I tilt the pen at different directions
- It has 13 buttons in total which are arranged as shown:
[image]

jamesbt365, 2022-04-18:
[M6.json.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/8505845/M6.json.txt)
 replace the configuration with this one, it should now have tilt in the debugger if you detect the tablet again, the input of the tablet should work correctly as well.

Stehsaer, 2022-04-18:
Everything works perfectly except buttons
[image]
12 out of 13 aux buttons are working, 1 missing
the touch wheel isn't working properly and outputs some combination of buttons when touched. it may be a good idea to temporarily disable it
does OTD support touch wheel?

Stehsaer, 2022-04-18:
the missing aux button isn't shown in the auxiliary button setting tab
there is only 12 shown
[image]

jamesbt365, 2022-04-18:
go back into the debugger, do specific actions with the wheel, get the raw data when it changes.

- What is it when not touched at all
- What is it when the button is pressed
- what happens when you scroll the wheel counter clockwise
- what happens when you scroll the wheel clockwise

Stehsaer, 2022-04-18:
The wheel doesn't output data saying whether it's rotating or not
Instead, it split itself into 12 different parts, with every part occupies 30 degrees on the wheel
These parts output a combination of aux buttons states when touched
This is how it works

When untouched:
[image]

When touched, simply split the whole wheel into 12 parts:
[image]
Look at the aux buttons states:
[image]
Consider the last 4 values as a REVERSED binary number ( False for 0, True for 1 )
In the example above, it's 0b1100, "12" in decimal.
It means that i'm touching the part labeled "12" in the first image.
This rule stays the same when touched the other parts.

It's somehow complicated and I have no idea whether there's existing code in OTD supporting this feature or not.

jamesbt365, 2022-04-18:
Can i have the raw data though, i would like to compare it to see if it fits along side a parser without breaking anything.

Stehsaer, 2022-04-18:
When touching part 1:
[image]
part 2:
[image]
part 3:
[image]
part 4:
[image]
part 5:
[image]

is these enough?
I found that the 6th byte in the raw data also represents the part no. touched

jamesbt365, 2022-04-18:
What about the button on on the wheel itself?

Stehsaer, 2022-04-18:
not clicked:
[image]
clicked:
[image]
it should be noticed that the last byte in the raw data changes between 01 and FF at a constant pace (about 2Hz) when the button is clicked

jamesbt365, 2022-04-18:
Can i have the raw data for every button when pressing individually please, starting from the wheel button and moving onwards through the rest.

Stehsaer, 2022-04-18:
Wheel Center Button:
[image]

12 buttons below the wheel, counting from the topleft corner, downwards to the bottomright corner
[image]

Button 1:
[image]
Button 2:
[image]
Button 3:
Button 4:
Button 4:

Stehsaer, 2022-04-18:
Like before, the last byte constantly changes when the button is hold
I guess it's an action of "forced refresh", and it may be ignored

jamesbt365, 2022-04-18:
When this action run is complete, download the corresponding build, extract it to its own folder and test the features of the tablet. The tablet should now have all 13 buttons. https://github.com/OpenTabletDriver/OpenTabletDriver/actions/runs/2184781618

jamesbt365, 2022-04-18:
@Stehsaer can you confirm or not?

Stehsaer, 2022-04-19:
> @Stehsaer can you confirm or not?

Yes, all feature are complete except the touch wheel.
The touch wheel false triggers other buttons when touched, which could cause some issues.
I suggest disabling the touch wheel

X9VoiD, 2022-04-19:
Test this build if the issue still exist

https://github.com/OpenTabletDriver/OpenTabletDriver/actions/runs/2192231712

InfinityGhost, 2022-05-03:
Pulling this from v0.6.0.4 so we can release.

jamesbt365, 2023-01-08:
#2203

PR #2203 (2022-04-18): Add support for Gaomon M6
fixes #2198
closed 2023-01-08 by jamesbt365