OpenTabletDriver/OpenTabletDriver #2200: Log messages have no limit on the amount of messages, leading to potentially high memory usage over time

gonX, 2022-04-18:
## Description
Writing a message to the console (via `Log`) fills up the RAM of OpenTabletDriver. Hypothetically, if a user were to leave the daemon running for long periods this would lead to very high memory usage.

It would probably be a good idea to hard cap this at 1000. This is unlikely to cause issues for diagnostic reports as we can just ask for a new one (lengthy ones are hard to read anyway). Being able to change this at startup via an environment variable might be interesting but probably not necessary.

Maybe we should have a variable internally that tells us when/if the log has been truncated? This would be useful to have added to diagnostics so we know whether its trimmed or not.

Realistically speaking this is not a huge issue in the current version of OpenTabletDriver, as we don't log _that_ verbosely, but custom plugins could in theory be causing issues here.

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         | All
| OpenTabletDriver Version | assuming 620fb42

jxu, 2022-05-06:
Is this the exported logfile from File > Export? Mine was 180 MB

jamesbt365, 2022-05-06:
Yes, this is exactly why.

jamesbt365, 2025-01-24:
Log files are now dumped to disk since #3100/#3108, capping log messages shouldn't be necessary (and realistically should be avoided anyway) as they are no longer stored in memory.

closed 2025-01-24 by jamesbt365