OpenTabletDriver/OpenTabletDriver #2206: opentabletdriver.service failing to start

ltsdw, 2022-04-20:
## Description

the service unit always fail to start.

I've been reading the [systemd opentabletdriver fails on system startup](https://github.com/OpenTabletDriver/OpenTabletDriver/discussions/1392), but even after the system is fully booted, trying to restart the service always result is the same.

increasing the `RestartSec` and `StartLimitBurst` only takes longer to fail. like if I set both to 300, it will take 5 minutes to fail.

I can run `otd` normally manually, and everything will just works, also running otd from startup with xinitrc file just works too. actually for me is preferable to run it this way, but I don't know why the service is failing if simply running `otd` just work.

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         |  Arch Linux x86_64
| OpenTabletDriver Version | 0.6.0.3-1
| Tablet                   | Wacom CTL-472

gonX, 2022-04-20:
Is your systemd user environment getting populated correctly?

This command should have some output:
`systemctl --user show-environment | grep DISPLAY`

Another relevant issue is likely [OpenTabletDriver.Web/43](https://github.com/OpenTabletDriver/OpenTabletDriver.Web/issues/43)

ltsdw, 2022-04-21:
I see.

so if you are starting your wm using startx there's no DISPLAY environment variable set yet before running startx, only after you run it.

only then you can call `systemctl --user import-environment DISPLAY XAUTHORITY`, preferably calling it inside xinitrc and not the script calling startx, didn't know that.

thank you!  closing.

closed 2022-04-21 by ltsdw