OpenTabletDriver/OpenTabletDriver #2209: Support Request: Huion H420 not detected, Grabbing existing conf from issue breaks daemon.

Left4Satan, 2022-04-22:
# Support Request

My Huion H420 is not detected, Grabbing existing conf from existing configuration request issue breaks daemon. or adding any existing confs to configuration folder breaks daemon. Had my friend Loki zip up his conf folder and it was errors gallore.
Tried to generate a config for it and 3 h420s show up and selecting one doesn't make any new conf. Disabling the device in device manager still allows the tablet to work after a pc restart we did as a troubleshooting measure even tho it's still disabled.
Also getting DeviceIOExceptions: 
[image]

And a JsonSerializationException:
[image]

## Diagnostic Information

Here is the first diag we exported before restarting the computer:
[DIAG1.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/8544363/DIAG1.zip)

 And another after restart made just before posting this:
[Diag2.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/8544365/Diag2.zip)

jamesbt365, 2022-04-22:
Yes, do not cross versions of configurations, configs from one version are not designed to work for another, of course it breaks. For the Huion H420, you need to install [zadig](https://github.com/pbatard/libwdi/releases/download/b755/zadig-2.6.exe) on interface 0.

closed 2022-04-22 by jamesbt365

TheFantasticLoki, 2022-04-23:
This shouldn't be closed yet. Satan is still having issues with his tablet resetting when his pen leaves the hover zone. I think it's something to do with the built in windows tablet drivers but I personally disabled all mouse pointer devices in device manager then only renabled the driver responsible for his mouse. This did Disable his tablet from working without OTD running untill we restarted the pc, The tablet continued to work without OTD open and with the device disabled in device manager under pointer devices. 

Can you give us some instructions on how he can remedy the issue?

jamesbt365, 2022-04-23:
Uninstall the oem drivers, and winusb of any interface other than 0.

TheFantasticLoki, 2022-04-23:
We already uninstalled OEM drivers BEFORE installing otd, and We only installed winusb on interface 0 like you instructed.
AND I had to disable the device in device manager because even without the oem drivers installed the tablet still functions similar to my own tablet to which I was instructed to disable the device through device manager in order to give OTD exclusive control over the device

jamesbt365, 2022-04-23:
help > export diagnostic, save that file and send it here, if i see WinUSB on any other interface you need to remove them, as i said above, disabling an interface does nothing, it needs to NOT has winusb on if it isn't interface 0.

TheFantasticLoki, 2022-04-23:
the diag2.zip contains an exported diagnostics report that was exported just before posting this thread. Satan is currently asleep but I will anydesk into his pc when he wakes up and manually uninstall winusb from any H420 listings aside from interface 0.

jamesbt365, 2022-04-23:
i need whats happening now, a diagnostic because anything was touched with 16 hours ago will not tell me what is happening now. Once i get a diagnostic i will be able to determine if its the system or otd, but from seeing no winusb in the diagnostics before i can only assume this was user error.

TheFantasticLoki, 2022-04-23:
Well I'll post an update when he wakes up and I can remote into his pc. But I did everything pertaining to this setup for him and the only thing we did after the post was install winusb on interface 0 which did allow otd to detect the tablet and start controlling it. I just got a message from him last night as I was in bed asking how to fix the cursor constantly teleporting to the top of the screen when he puts his pen on his tablet. I'll see if I can get him to record a video of it happening too cause I wasn't entirely sure what he was talking about myself.

jamesbt365, 2022-04-23:
Cursor teleportation indicates some of the below:

- Other drivers are getting in the way
- There is a lot of devices around which cause interference to the tablet
- Pen buttons are being pressed (applies to anything other than wacom because of the tablet firmware)
- The parser is incorrect (unlikely because huion probably wouldn't "update" one of there old tablets that are dirt cheap and would probably break a bit more than teleporting when going into range)
- or if this is specifically an application like osu!, "map absolute raw input to the osu! window" must be turned off and sensitivity must be 1 due to the nature of absolute devices.

TheFantasticLoki, 2022-04-23:
> 

I believe it's likely scenario 1 but will investigate that when I can.

Can you elaborate on Scenario 2 as I've never heard of this happening unless the device has a sufficient magnet in it like a phone case magnet or something.

For scenario 3 would disabling the button binds in otd fix this or nah cause I already cleared his pen button bindings except for the pen nib.

And for scenario 5 I already double checked these settings and told him to NEVER EVER CHANGE THEM from sens 1x and never to turn on raw input or map absolute raw input.

jamesbt365, 2022-04-23:
scenario 2, just some devices spit out alot of EMR which causes stuff, though usually isn't an issue as the pen is shielded as so is most of the devices actually emitting it.

Scenario 3, thats a firmware issue, and there is no way around it, no driver can fix it, its literally just the tablet itself overriding its own inputs.

Left4Satan, 2022-04-23:
Here is the freshly exported diagnostics report since he woke up:
[DIAG3.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/8547420/DIAG3.zip)

And here is a small (Very scatterbrianed) clip showcasing some of the NEW issues that just showed up today:
https://streamable.com/knk5m3

jamesbt365, 2022-04-23:
FYI: if you listened to what i said in the first message, you can't have a configurations folder from another version, stuff changes. Delete it.

Left4Satan, 2022-04-23:
We already did delete it the confs that are present were made using the otd on this machine while testing but we will remove it just in case

jamesbt365, 2022-04-23:
Also OTD shouldn't be on onedrive.

Left4Satan, 2022-04-23:
OMFG I HATE ONEDRIVE SO FUCKING MUCH WHY IS IT FORCED ON BY DEFAULT Flkfjlfjalfjasl;fjasklo

jamesbt365, 2022-04-23:
i feel that.

jamesbt365, 2022-04-23:
Anyway the only reason otd is failing now is because of the `Configurations` folder. And OneDrive being annoying, other than that everything looks fine. The debugger is freezing because the tablet configuration is not what it thinks it should be and therefore the entire configuration fails including closing the device which is why the tablet drops back into windows ink and pnp

TheFantasticLoki, 2022-04-23:
Alright I think we are good now... His Quote "Issue" was that he thought the tablet functioned like a mouse so relative positioning not absolute and he was confused but his phrasing was dogwater. 

Moved the otd binary location to it's own folder in localappdata and he has proper control again. Thank you for dealing with this absolute waste of time we put you through cause we're idiots...

jamesbt365, 2022-04-23:
No problem, have a nice day, and hope everything else works for you.