OpenTabletDriver/OpenTabletDriver #2211: Support Request:

DrunkenMasterTim, 2022-04-24:
YOU GUYS KEEP TELLING ME TO STOP OPENING SUPPORT TICKETS FOR THE SAME ISSUE YET MY ISSUE HAS NOT BEEN RESOLVEED ITS BEEN WEEKS AND YOU CLOSED MY TICKET, THEN TELL ME TO GO TO THE DISCORD WHAT IS THE GODDAMN DISCORD THEN, ALL I WANNA DO IS PLAY OSU

jamesbt365, 2022-04-24:
opentabletdriver.net/Discord

closed 2022-04-24 by jamesbt365

InfinityGhost, 2022-04-24:
Creating multiple issues will not help your problem. Stick to the original issue (#2136) and don't create any more issues relating to this problem, otherwise join the [Discord](https://opentabletdriver.net/Discord) for more direct support.

Keep in mind that everyone who helps you here is doing this out of their own free will and out of the kindness of their own hearts, we can't keep up with everyone at all times. This poor behavior will not be tolerated and any more outbursts as such will result in you being blocked from creating issues or participating in discussions.

Locking this conversation, continue in the issue mentioned above.