OpenTabletDriver/OpenTabletDriver #2214: Support Request: Tablet cursor acting weird when using tablet

taigochan, 2022-04-24:
It's hard to explain so i also made a video but i'll try my best.
My cursor is moving from one to another side of the screen which and when i lift it up it appears on a different side of the screen.
I think this is because another driver is working in the background and the console also displayed "[Detect:Warning]	Another tablet driver found: Wacom", however i don't have any other drivers installed so i think it might be built in with the plasma-desktop or something else. It seemed to do this in Absolute mode and Artist mode, in relative mode it resets the settings so its probally using the same settings as that mysterious driver in the background and therefore the problem doesn't exsist.

here's the video: https://streamable.com/rpnpbj

## What I've already tried
I tried searching for wiki's or any drivers that i might have installed somehow. i also searched through issues but didn't find something there so im writing it now.

## Diagnostic Information
[diagnostics.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/8550147/diagnostics.txt)

## Other info
tablet: Wacom CTL-472
notes: This happened on a pretty fresh install and am absolutely certain i have no other drivers installed. I installed arch linux and with it i installed "sddm" and "plasma-desktop" packages

If there's anything else you need to know you need only to ask :)

jamesbt365, 2022-04-24:
`rmmod wacom`

taigochan, 2022-04-24:
uhh, wow that was quick... and it works! thank you very much

closed 2022-04-24 by taigochan

jamesbt365, 2022-04-24:
http://opentabletdriver.net/Wiki/FAQ/Linux#ghost-cursor for perm solution.