OpenTabletDriver/OpenTabletDriver #2221: Trying to use the rpm package found in releases, doesnt work and tries to look for mono

Sakooooo, 2022-05-01:
```
Error: 
 Problem: conflicting requests
  - nothing provides mono(Microsoft.Win32.Primitives) = 4.1.1.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(Microsoft.Win32.Primitives) = 5.0.0.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(Microsoft.Win32.Registry) = 4.1.2.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(Newtonsoft.Json) = 12.0.0.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Collections) = 4.1.1.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Collections) = 4.1.2.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Collections) = 5.0.0.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Collections) = 6.0.0.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Collections.Concurrent) = 4.0.14.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Collections.Concurrent) = 6.0.0.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Collections.Immutable) = 6.0.0.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.ComponentModel) = 6.0.0.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.ComponentModel.Primitives) = 6.0.0.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Console) = 6.0.0.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Diagnostics.Debug) = 4.1.1.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Diagnostics.Debug) = 4.1.2.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Diagnostics.Process) = 4.2.1.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Diagnostics.Process) = 6.0.0.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Diagnostics.Tools) = 4.1.1.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Diagnostics.Tools) = 4.1.2.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Diagnostics.TraceSource) = 4.1.1.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Diagnostics.TraceSource) = 4.1.2.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Diagnostics.TraceSource) = 5.0.0.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Diagnostics.Tracing) = 6.0.0.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.IO.Compression) = 6.0.0.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.IO.Compression.ZipFile) = 6.0.0.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.IO.FileSystem) = 5.0.0.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.IO.Pipes) = 4.1.2.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.IO.Pipes) = 6.0.0.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Linq) = 4.2.1.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Linq) = 4.2.2.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Linq) = 5.0.0.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Linq) = 6.0.0.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Linq.Expressions) = 4.2.1.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Linq.Expressions) = 6.0.0.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Linq.Parallel) = 6.0.0.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Memory) = 4.1.0.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Memory) = 4.2.0.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Memory) = 4.2.1.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Memory) = 6.0.0.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Net.Http) = 4.2.1.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Net.Http) = 6.0.0.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Net.Primitives) = 4.1.2.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Net.Primitives) = 6.0.0.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Net.Sockets) = 4.2.2.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Net.WebSockets) = 4.1.1.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Net.WebSockets) = 4.1.2.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Numerics.Vectors) = 6.0.0.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.ObjectModel) = 4.1.1.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.ObjectModel) = 6.0.0.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Reflection.Emit) = 4.1.1.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Reflection.Emit.ILGeneration) = 4.0.3.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Reflection.Emit.ILGeneration) = 6.0.0.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Reflection.Emit.Lightweight) = 4.0.3.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Reflection.Emit.Lightweight) = 6.0.0.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Reflection.Primitives) = 4.1.1.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Reflection.Primitives) = 6.0.0.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Resources.ResourceManager) = 4.1.1.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Resources.ResourceManager) = 4.1.2.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Runtime) = 4.2.1.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Runtime) = 4.2.2.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Runtime) = 5.0.0.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Runtime) = 6.0.0.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Runtime.CompilerServices.Unsafe) = 6.0.0.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Runtime.Extensions) = 4.2.1.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Runtime.Extensions) = 4.2.2.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Runtime.InteropServices) = 4.2.1.0 needed by opentabletdriver-0.6.0.3-1.x86_64
[... 13 lines cut ...]
  - nothing provides mono(System.Text.Json) = 6.0.0.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Text.RegularExpressions) = 4.2.1.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Text.RegularExpressions) = 5.0.0.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Text.RegularExpressions) = 6.0.0.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Threading) = 4.1.1.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Threading) = 4.1.2.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Threading) = 5.0.0.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Threading) = 6.0.0.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Threading.Overlapped) = 5.0.0.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Threading.Overlapped) = 6.0.0.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Threading.Tasks) = 4.1.1.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Threading.Tasks) = 4.1.2.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Threading.Tasks.Dataflow) = 4.6.5.0 needed by opentabletdriver-0.6.0.3-1.x86_64
[... cut ...]
  - nothing provides mono(System.Threading.ThreadPool) = 5.0.0.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Threading.ThreadPool) = 6.0.0.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Threading.Timer) = 4.1.1.0 needed by opentabletdriver-0.6.0.3-1.x86_64
  - nothing provides mono(System.Xml.XDocument) = 4.1.1.0 needed by opentabletdriver-0.6.0.3-1.x86_64
(try to add '--skip-broken' to skip uninstallable packages)

```
package tries to look for mono, any fix? ive tried adding the mono repo to fedora still no change

jamesbt365, 2022-05-01:
OpenTabletDriver/OpenTabletDriver.Packaging/pull/16

closed 2022-05-01 by jamesbt365

m2jest1c, 2022-11-25:
As for a quick hotfix using the existing RPMs in the existing releases, using `sudo rpm -i --nodeps OpenTabletDriver.rpm` works fine, as long as you have some sort of mono installed. Upgrading between Fedora releases requires you to redo this, though.