OpenTabletDriver/OpenTabletDriver #2225: Scrolling with tablet seems impossible in games

hypixus, 2022-05-04:
## Description

The drivers do not allow for proper scroll movement in Osu! game.
I am certain issue is with the driver, because of fact this did never happen while using Hawku's driver.
Anytime I attempt to scroll the song selection, scroll action works as long as no movement is made while the pen nib is no longer pressed, for example
1. Press pen down
2. Swiftly move pen downwards
3. Release pen while still moving
4. Lift up pen entirely
At step 3, it usually goes back to the initial position or even scrolls backwards. This bug does not occur with old hawku driver.

Here is the configuration:
[image]

No settings of buttons were changed, nor any plugins were enabled.

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         | Windows 10 21H2
| OpenTabletDriver Version | 0.6.0.3
| Tablet                   | Wacom CTL-490

jamesbt365, 2022-05-04:
CTL-490 hardware issue.

hypixus, 2022-05-06:
Apologies for late response.

[diagfile.json.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/8640053/diagfile.json.txt)

Renamed the ending to txt because github cried. I tried the recorder you asked for, yet I cannot get where does it save the results. Diagfile was ~100 lines more after it, so I hope this is it.

jamesbt365, 2022-05-06:
Wacom Tablets do not use EMA smoothing, therefore reconstructor is not suitable for these tablets. I'm also still missing a recording, could you send one?

hypixus, 2022-05-06:
Certainly, here is a [link](https://streamable.com/04ik7y) to the video.

Here I'm trying to move along the song selection list upwards with opposite result, and the other way around. Finally, I move my cursor to the song title which resets the song select position.
If I had to describe it acts like a slingshot - pulling it down results in springing back upwards, despite never messing with the driver nor game acting this way with old drivers. This was tested multiple times.

jamesbt365, 2022-05-06:
Seen here is your cursor teleporting for 1 report. This is either because you are using reconstructor on a tablet that doesn't use EMA smoothing, thus  causing problems because it isn't removing the right thing or because the x90 wacom tablets have hover smoothing and bad firmware causing weird interactions when the pen is lifted.
[image]

hypixus, 2022-05-06:
This is bad news... Is there any settings I could change to prevent that/try fixing it? As mentioned, I do not use any plugins or have any of the defaults changed from lastest github release.

jamesbt365, 2022-05-06:
The diagnostic shows that you have reconstructor on, or at least did.

hypixus, 2022-05-06:
My bad, forgot to turn it off. However, after uninstalling it, I got the same result, despite running the driver again.

Kuuuube, 2022-05-06:
This is a known hardware issue with the CTx-x90 tablets. Using BezierInterpolator filter on default settings should help mitigate this without much issue.

hypixus, 2022-05-06:
It does help a tiny bit, yet still - it is a flaw in the product, not your hard work.

Apologies and thanks for your time, closing the issue.