OpenTabletDriver/OpenTabletDriver #2228: Applying WinInk settings via tablet input fails with a NullReferenceException

Sinamer, 2022-05-05:
## Description
Pressing Save/Apply or switching to a WinInk preset by using a tablet input fails.
The issue doesn't happen if this is done with a mouse input instead.

Relevant Error log in Console:
`[NullReferenceException:Error] Object reference not set to an instance of an object.`
`at OpenTabletDriver.Plugin.Output.AbsoluteOutputMode.OnOutput(IDeviceReport report)`
``at OpenTabletDriver.Devices.DeviceReader`1.Main()``

And a Debug message:
`[System.InvalidOperationException:Debug]    Collection was modified; enumeration operation may not execute.`
  `` at System.Collections.Generic.List`1.Enumerator.MoveNextRare()``
   `at OpenTabletDriver.Daemon.DriverDaemon.SetSettings(Settings settings)`

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System | Windows 10 build 17763        |
| OpenTabletDriver Version | 0.6.0.4 |
| Tablet | XP-Pen Star G430                  |

jamesbt365, 2022-06-18:
Wrong repository for reporting issues with plugins, they should be reported to the respective repository though this was likely fixed with a hotfix to windows ink.

Sinamer, 2022-06-18:
Sorry if this is wrong, but from a short [discord message](https://discord.com/channels/615607687467761684/628651971267788800/971901528224976946) from InfinityGhost when this issue was created, it sounds like something that should be handled on OTD's side?

jamesbt365, 2022-06-18:
oh i misread the title this is because you are doing it with pen, my bad. Will reopen

gonX, 2023-03-28:
Fixed by #2552

Sinamer, 2023-12-29:
Is this issue supposed to be fixed in v0.6.4.0 (latest OTD release)?
Still seeing this:

https://github.com/OpenTabletDriver/OpenTabletDriver/assets/53346335/400c9bfe-01ca-4910-8e74-b24afda9c90a

jamesbt365, 2023-12-29:
This won't really be fixed entirely until 0.7 hits I think, just use your mouse for now.

PR #2476 (2022-11-14): Resolve race condition bugs of Driver.InputDevices
Use immutable array and atomically update InputDevices.

Fixes ppy/osu#21078
Fixes #1718
Fixes #2228