OpenTabletDriver/OpenTabletDriver #2229: Ubuntu Huion 420 ArgumentOutOfRangeException: Value range is [0, 15]. (Parameter 'value')

jxu, 2022-05-06:
## Description
Error while running inside osu lazer. runtime.log

```
2022-05-06 05:44:01 [verbose]: ----------------------------------------------------------
2022-05-06 05:44:01 [verbose]: runtime Log for jx (LogLevel: Verbose)
2022-05-06 05:44:01 [verbose]: Running osu 2022.501.0.0 on .NET 6.0.2
2022-05-06 05:44:01 [verbose]: Environment: Linux (Unix 5.4.0.99), 8 cores
2022-05-06 05:44:01 [verbose]: ----------------------------------------------------------
2022-05-06 05:44:01 [verbose]: Detect: Searching for tablets...
2022-05-06 05:44:01 [error]: ArgumentOutOfRangeException: Value range is [0, 15]. (Parameter 'value')
2022-05-06 05:44:01 [error]: ArgumentOutOfRangeException: Value range is [0, 15]. (Parameter 'value')
2022-05-06 05:44:01 [error]: ArgumentOutOfRangeException: Value range is [0, 15]. (Parameter 'value')
2022-05-06 05:44:01 [error]: ArgumentOutOfRangeException: Value range is [0, 15]. (Parameter 'value')
2022-05-06 05:44:01 [error]: ArgumentOutOfRangeException: Value range is [0, 15]. (Parameter 'value')
2022-05-06 05:44:01 [error]: ArgumentOutOfRangeException: Value range is [0, 15]. (Parameter 'value')
2022-05-06 05:44:01 [error]: ArgumentOutOfRangeException: Value range is [0, 15]. (Parameter 'value')
2022-05-06 05:44:01 [error]: ArgumentOutOfRangeException: Value range is [0, 15]. (Parameter 'value')
2022-05-06 05:44:01 [error]: ArgumentOutOfRangeException: Value range is [0, 15]. (Parameter 'value')
2022-05-06 05:44:01 [important]: Detect: More than 1 matching device has been found.
2022-05-06 05:44:01 [error]: ArgumentOutOfRangeException: Value range is [0, 15]. (Parameter 'value')
2022-05-06 05:44:01 [verbose]: Detect: Found tablet 'Huion 420'
2022-05-06 05:44:01 [error]: ArgumentOutOfRangeException: Value range is [0, 15]. (Parameter 'value')
2022-05-06 05:44:02 [error]: ArgumentOutOfRangeException: Value range is [0, 15]. (Parameter 'value')
2022-05-06 05:44:02 [error]: ArgumentOutOfRangeException: Value range is [0, 15]. (Parameter 'value')
2022-05-06 05:44:02 [error]: ArgumentOutOfRangeException: Value range is [0, 15]. (Parameter 'value')
2022-05-06 05:44:02 [error]: ArgumentOutOfRangeException: Value range is [0, 15]. (Parameter 'value')
2022-05-06 05:44:02 [error]: ArgumentOutOfRangeException: Value range is [0, 15]. (Parameter 'value')
2022-05-06 05:44:02 [error]: ArgumentOutOfRangeException: Value range is [0, 15]. (Parameter 'value')
2022-05-06 05:44:02 [error]: ArgumentOutOfRangeException: Value range is [0, 15]. (Parameter 'value')
2022-05-06 05:44:02 [error]: ArgumentOutOfRangeException: Value range is [0, 15]. (Parameter 'value')
2022-05-06 05:44:02 [error]: ArgumentOutOfRangeException: Value range is [0, 15]. (Parameter 'value')
2022-05-06 05:44:02 [error]: ArgumentOutOfRangeException: Value range is [0, 15]. (Parameter 'value')
2022-05-06 05:44:02 [error]: ArgumentOutOfRangeException: Value range is [0, 15]. (Parameter 'value')
2022-05-06 05:44:02 [error]: ArgumentOutOfRangeException: Value range is [0, 15]. (Parameter 'value')
2022-05-06 05:44:02 [error]: ArgumentOutOfRangeException: Value range is [0, 15]. (Parameter 'value')
2022-05-06 05:44:02 [error]: ArgumentOutOfRangeException: Value range is [0, 15]. (Parameter 'value')
2022-05-06 05:44:02 [error]: ArgumentOutOfRangeException: Value range is [0, 15]. (Parameter 'value')
2022-05-06 05:44:02 [error]: ArgumentOutOfRangeException: Value range is [0, 15]. (Parameter 'value')
2022-05-06 05:44:02 [error]: ArgumentOutOfRangeException: Value range is [0, 15]. (Parameter 'value')
2022-05-06 05:44:03 [error]: ArgumentOutOfRangeException: Value range is [0, 15]. (Parameter 'value')
2022-05-06 05:44:03 [error]: ArgumentOutOfRangeException: Value range is [0, 15]. (Parameter 'value')
2022-05-06 05:44:03 [error]: ArgumentOutOfRangeException: Value range is [0, 15]. (Parameter 'value')
2022-05-06 05:44:03 [error]: ArgumentOutOfRangeException: Value range is [0, 15]. (Parameter 'value')
2022-05-06 05:44:03 [error]: ArgumentOutOfRangeException: Value range is [0, 15]. (Parameter 'value')
2022-05-06 05:44:03 [error]: ArgumentOutOfRangeException: Value range is [0, 15]. (Parameter 'value')
2022-05-06 05:44:03 [error]: ArgumentOutOfRangeException: Value range is [0, 15]. (Parameter 'value')
2022-05-06 05:44:03 [error]: ArgumentOutOfRangeException: Value range is [0, 15]. (Parameter 'value')
2022-05-06 05:44:03 [error]: ArgumentOutOfRangeException: Value range is [0, 15]. (Parameter 'value')
2022-05-06 05:44:03 [error]: ArgumentOutOfRangeException: Value range is [0, 15]. (Parameter 'value')
2022-05-06 05:44:03 [error]: ArgumentOutOfRangeException: Value range is [0, 15]. (Parameter 'value')
2022-05-06 05:44:03 [error]: ArgumentOutOfRangeException: Value range is [0, 15]. (Parameter 'value')
2022-05-06 05:44:03 [error]: ArgumentOutOfRangeException: Value range is [0, 15]. (Parameter 'value')
2022-05-06 05:44:03 [error]: ArgumentOutOfRangeException: Value range is [0, 15]. (Parameter 'value')
2022-05-06 05:44:03 [error]: ArgumentOutOfRangeException: Value range is [0, 15]. (Parameter 'value')
2022-05-06 05:44:03 [error]: ArgumentOutOfRangeException: Value range is [0, 15]. (Parameter 'value')
2022-05-06 05:44:04 [error]: ArgumentOutOfRangeException: Value range is [0, 15]. (Parameter 'value')
2022-05-06 05:44:04 [error]: ArgumentOutOfRangeException: Value range is [0, 15]. (Parameter 'value')
2022-05-06 05:44:04 [error]: ArgumentOutOfRangeException: Value range is [0, 15]. (Parameter 'value')
2022-05-06 05:44:04 [error]: ArgumentOutOfRangeException: Value range is [0, 15]. (Parameter 'value')
2022-05-06 05:44:04 [error]: ArgumentOutOfRangeException: Value range is [0, 15]. (Parameter 'value')
2022-05-06 05:44:04 [error]: ArgumentOutOfRangeException: Value range is [0, 15]. (Parameter 'value')
```

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         | Ubuntu 20.04
| OpenTabletDriver Version | Latest (0.6.04)
| Tablet                   | Huion 420

SnowLire, 2022-05-06:
The solution is present on the [Linux FAQ page](http://opentabletdriver.net/Wiki/FAQ/Linux#argumentoutofrangeexception).

jxu, 2022-05-06:
I tried rmmod and had the same issue. Reopen the issue

jamesbt365, 2022-05-06:
Before we reopen does this happen on standalone otd with the module removed? Also check if the module is still running `lsmod | grep hid_uclogic` if get output then its still running.

jxu, 2022-05-06:
> Before we reopen does this happen on standalone otd with the module removed? Also check if the module is still running `lsmod | grep hid_uclogic` if get output then its still running.

Wdym standalone? Running as a service or with the GUI?  

I have tried using rmmod again, so lsmod doesn't report hid_uclogic. osu ingame detects the tablet but it does not respond.

```
2022-05-06 22:28:44 [important]: Detect: More than 1 matching device has been found.
2022-05-06 22:28:44 [verbose]: Detect: Found tablet 'Huion 420'
```

Doesn't show up in xinput:

```
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ SynPS/2 Synaptics TouchPad              	id=19	[slave  pointer  (2)]
⎜   ↳ TPPS/2 IBM TrackPoint                   	id=20	[slave  pointer  (2)]
⎜   ↳ Corsair Corsair K65 Gaming Keyboard     	id=14	[slave  pointer  (2)]
⎜   ↳ USB Optical Mouse                       	id=23	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Video Bus                               	id=7	[slave  keyboard (3)]
    ↳ Video Bus                               	id=8	[slave  keyboard (3)]
    ↳ Sleep Button                            	id=9	[slave  keyboard (3)]
    ↳ Integrated Camera: Integrated C         	id=17	[slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard            	id=18	[slave  keyboard (3)]
    ↳ ThinkPad Extra Buttons                  	id=21	[slave  keyboard (3)]
    ↳ Corsair Corsair K65 Gaming Keyboard     	id=15	[slave  keyboard (3)]
    ↳ Corsair Corsair K65 Gaming Keyboard     	id=16	[slave  keyboard (3)]
    ↳ Corsair Corsair K65 Gaming Keyboard     	id=22	[slave  keyboard (3)]
```

jamesbt365, 2022-05-06:
by standalone otd i don't mean in lazer, just run otd normally and not in lazer.

jxu, 2022-05-06:
I do have the daemon running if that's what you mean.

I opened the GTK UX and the tablet shows up and all, but does not actually respond to input

```
[Detect:Debug]	Searching for tablet 'XP-Pen Artist 12'
[Detect:Debug]	Searching for tablet 'XP-Pen Artist 13.3 Pro'
[Detect:Debug]	Searching for tablet 'XP-Pen Artist 15.6 Pro'
[Detect:Debug]	Searching for tablet 'XP-Pen Artist 15.6'
[Detect:Debug]	Searching for tablet 'XP-Pen Artist 22HD'
[Detect:Debug]	Searching for tablet 'XP-Pen CT1060'
[Detect:Debug]	Searching for tablet 'XP-Pen CT430'
[Detect:Debug]	Searching for tablet 'XP-Pen CT640'
[Detect:Debug]	Searching for tablet 'XP-Pen Deco 01 V2 (variant 2)'
[Detect:Debug]	Searching for tablet 'XP-Pen Deco 01 V2'
[Detect:Debug]	Searching for tablet 'XP-Pen Deco 01'
[Detect:Debug]	Searching for tablet 'XP-Pen Deco 02'
[Detect:Debug]	Searching for tablet 'XP-Pen Deco 03'
[Detect:Debug]	Searching for tablet 'XP-Pen Deco mini4'
[Detect:Debug]	Searching for tablet 'XP-Pen Deco mini7'
[Detect:Debug]	Searching for tablet 'XP-Pen Deco Pro Medium'
[Detect:Debug]	Searching for tablet 'XP-Pen Deco Pro Small'
[Detect:Debug]	Searching for tablet 'XP-Pen Innovator 16'
[Detect:Debug]	Searching for tablet 'XP-Pen Star 03'
[Detect:Debug]	Searching for tablet 'XP-Pen Star 05 V3'
[Detect:Debug]	Searching for tablet 'XP-Pen Star 06'
[Detect:Debug]	Searching for tablet 'XP-Pen Star 06C'
[Detect:Debug]	Searching for tablet 'XP-Pen Star G430'
[Detect:Debug]	Searching for tablet 'XP-Pen Star G430S V2'
[Detect:Debug]	Searching for tablet 'XP-Pen Star G430S'
[Detect:Debug]	Searching for tablet 'XP-Pen Star G540 Pro'
[Detect:Debug]	Searching for tablet 'XP-Pen Star G540'
[Detect:Debug]	Searching for tablet 'XP-Pen Star G640 V2'
[Detect:Debug]	Searching for tablet 'XP-Pen Star G640'
[Detect:Debug]	Searching for tablet 'XP-Pen Star G640S'
[Detect:Debug]	Searching for tablet 'XP-Pen Star G960'
[Detect:Debug]	Searching for tablet 'XP-Pen Star G960S Plus'
[Detect:Debug]	Searching for tablet 'XP-Pen Star G960S'
[Tablet:Debug]	Tablet debugging is enabled
[Tablet:Debug]	Tablet debugging is disabled
[Huion 420:Info]	Output mode: Absolute Mode
[Huion 420:Info]	Display area: [3840x1080@<1920, 540>:0°],
[Huion 420:Info]	Tablet area: [101.6x57.01295@<50.8, 28.506475>:0°],
[Huion 420:Info]	Clipping: Enabled
[Huion 420:Info]	Ignoring reports outside area: Disabled
[Huion 420:Info]	Tip Binding: [Mouse Button Binding: Left]@0%
[Settings:Info]	Driver is enabled.
[Detect:Info]	Searching for tablets...
[Device:Debug]	A device stream was disposed.
[Detect:Debug]	Searching for tablet '10moon 1060N'
[Detect:Debug]	Searching for tablet 'Acepen AP 1060'
[Detect:Debug]	Searching for tablet 'Artisul A1201'
[Detect:Debug]	Searching for tablet 'Artisul AP604'
[Detect:Debug]	Searching for tablet 'Artisul M0610 Pro'
[Detect:Debug]	Searching for tablet 'Gaomon 1060 Pro'
[Detect:Debug]	Searching for tablet 'Gaomon M106K Pro'
[Detect:Debug]	Searching for tablet 'Gaomon M106K'
[Detect:Debug]	Searching for tablet 'Gaomon M10K Pro'
[Detect:Debug]	Searching for tablet 'Gaomon M10K'
[Detect:Debug]	Searching for tablet 'Gaomon M1220'
[Detect:Debug]	Searching for tablet 'Gaomon M1230'
[Detect:Debug]	Searching for tablet 'Gaomon M6'
[Detect:Debug]	Searching for tablet 'Gaomon PD1161'
[Detect:Debug]	Searching for tablet 'Gaomon PD1560'
[Detect:Debug]	Searching for tablet 'Gaomon PD1561'
[Detect:Debug]	Searching for tablet 'Gaomon S56K'
[Detect:Debug]	Searching for tablet 'Gaomon S620'
[Detect:Debug]	Searching for tablet 'Gaomon S630'
[Detect:Debug]	Searching for tablet 'Genius G-Pen 560'
[Detect:Debug]	Searching for tablet 'Genius i608x'
[... 161 lines cut ...]
[Detect:Debug]	Searching for tablet 'XP-Pen Star 03'
[Detect:Debug]	Searching for tablet 'XP-Pen Star 05 V3'
[Detect:Debug]	Searching for tablet 'XP-Pen Star 06'
[Detect:Debug]	Searching for tablet 'XP-Pen Star 06C'
[Detect:Debug]	Searching for tablet 'XP-Pen Star G430'
[Detect:Debug]	Searching for tablet 'XP-Pen Star G430S V2'
[Detect:Debug]	Searching for tablet 'XP-Pen Star G430S'
[Detect:Debug]	Searching for tablet 'XP-Pen Star G540 Pro'
[Detect:Debug]	Searching for tablet 'XP-Pen Star G540'
[Detect:Debug]	Searching for tablet 'XP-Pen Star G640 V2'
[Detect:Debug]	Searching for tablet 'XP-Pen Star G640'
[Detect:Debug]	Searching for tablet 'XP-Pen Star G640S'
[Detect:Debug]	Searching for tablet 'XP-Pen Star G960'
[Detect:Debug]	Searching for tablet 'XP-Pen Star G960S Plus'
[Detect:Debug]	Searching for tablet 'XP-Pen Star G960S'
[Huion 420:Info]	Output mode: Absolute Mode
[Huion 420:Info]	Display area: [3840x1080@<1920, 540>:0°],
[Huion 420:Info]	Tablet area: [101.6x57.01295@<50.8, 28.506475>:0°],
[Huion 420:Info]	Clipping: Enabled
[Huion 420:Info]	Ignoring reports outside area: Disabled
[Huion 420:Info]	Tip Binding: [Mouse Button Binding: Left]@0%
[Settings:Info]	Driver is enabled.
[IPC:Debug]	Disconnected from a client.
[IPC:Debug]	Connected to a client.
```

jamesbt365, 2022-05-06:
help > export diagnostics, save that file and send it here.

jxu, 2022-05-06:
The exported logfile is huge so here it is compressed. Maybe multiple input devices are open at once? 

[export.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/8643945/export.zip)

jamesbt365, 2022-05-06:
Something is fucked up with your USB Stack, if rebooting doesn't fix it then you're for the most part on your own.

jxu, 2022-05-06:
Ok I rebooted and it fixed itself. Idk what the issue was.

jxu, 2022-05-07:
It appears every time the tablet is reconnected, a new device is created instead of closing the old device. So when osu opens, it has some race condition to choose which device.

jxu, 2022-05-07:
This is probably why it says "more than 1 matching device has been found". Is there a way to close all hidraw devices?

[export2.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/8644541/export2.txt)

gonX, 2022-05-08:
Reopening as this is an ongoing support issue

jamesbt365, 2022-05-08:
This relevant?
https://github.com/OpenTabletDriver/OpenTabletDriver/pull/2133#issuecomment-1089276414

Well the reason it matches 2 devices is because there is 2 8, 0, 0 identifiers not sure how we would get around that.

jxu, 2022-05-08:
Is there a way to manually list and close sys devices? That might solve the problem although idk the hidraw api

jamesbt365, 2023-08-17:
Either this issue should be renamed to something that fits the underlying problem or a new issue should be opened about matching identifiers causing conflicts (only one identifier has data so its first come from serve for whatever otd opens)

jxu, 2023-08-18:
If I play Lazer again I'll see if the issue is present

jamesbt365, 2023-08-18:
This is tracking another problem.

kostek001, 2023-12-14:
I found temporary "fix"
It involves changing udev rules, to only allow specified interface number
```
SUBSYSTEMS=="usb", ENV{ID_USB_INTERFACE_NUM}="$attr{bInterfaceNumber}"

SUBSYSTEM=="hidraw", ATTRS{idVendor}=="256c", ATTRS{idProduct}=="006f", ENV{ID_USB_INTERFACE_NUM}=="00", MODE="0666"
SUBSYSTEM=="usb", ATTRS{idVendor}=="256c", ATTRS{idProduct}=="006f", MODE="0666"
SUBSYSTEM=="input", ATTRS{idVendor}=="256c", ATTRS{idProduct}=="006f", ENV{ID_USB_INTERFACE_NUM}=="00", ENV{LIBINPUT_IGNORE_DEVICE}="1"
```

X9VoiD, 2023-12-15:
@kostek001 just update your opentabletdriver install to 0.6.3.0 and follow instructions [here](https://opentabletdriver.net/Wiki/Documentation/RequiredPermissions#linux). This issue is tracking something else. Will rename this issue to reflect that.

kostek001, 2023-12-15:
@X9VoiD
No this is a correct issue.

Huion HS611 has 3 endpoints.
Output of `lsusb -v`:
```
Bus 001 Device 103: ID 256c:006f HUION Huion Tablet_HS611
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x256c HUION
  idProduct          0x006f Huion Tablet_HS611
  bcdDevice            1.12
  iManufacturer           1 HUION
  iProduct                2 Huion Tablet_HS611
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              480mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      18
          Report Descriptors: 
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               2
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      86
          Report Descriptors: 
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               2
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 [unknown]
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      67
          Report Descriptors: 
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
Device Status:     0x0002
  (Bus Powered)
  Remote Wakeup Enabled
```

I've modified udev rules to only set rw permissions for the first endpoint, so OTD cannot connect to other two.
```
# Adds a variable with the interface number
SUBSYSTEMS=="usb", ENV{ID_USB_INTERFACE_NUM}="$attr{bInterfaceNumber}"

# Huion HS611
SUBSYSTEM=="hidraw", ATTRS{idVendor}=="256c", ATTRS{idProduct}=="006f", ENV{ID_USB_INTERFACE_NUM}=="00", MODE="0666"
SUBSYSTEM=="usb", ATTRS{idVendor}=="256c", ATTRS{idProduct}=="006f", MODE="0666"
SUBSYSTEM=="input", ATTRS{idVendor}=="256c", ATTRS{idProduct}=="006f", ENV{ID_USB_INTERFACE_NUM}=="00", ENV{LIBINPUT_IGNORE_DEVICE}="1"
```

X9VoiD, 2023-12-15:
Right, but it still stands that you should update to the latest OTD because you're using an old variation of the udev rules. On later OTD versions, it's using `udev-acl` and `uaccess` instead of allowing everything to read the input device.

That fix will also not work on 0.7.x.

There is progress however in #2772 and #3011.

gonX, 2024-11-23:
I think this is fixed with #3245 / #3475?

Tablets that still aren't matched correctly should make a new issue.

The last remaining parts (if any) that is needed to be implemented in OpenTabletDriver should be addressed in #3011.

closed 2024-11-23 by gonX