OpenTabletDriver/OpenTabletDriver #2231: Debian 11: dpkg error and uninstallable dependancies

Manga-Man, 2022-05-06:
# Support Request

Hewwo~ , Im using debian 11 and I keep getting 'E: Sub-process /usr/bin/dpkg returned an error code (1)' whenever I run the sudo apt install -y apt-transport-https command.
Ive tried skipping that error but I get uninstallable dependancy error:

[image]
[image]

## What I've already tried
Ive tried common things like force install and reconfiguring.

jamesbt365, 2022-05-06:
Shouldn't explicitly be a packaging issue, many people use the .deb on all sorts of versions, this just is a conflicting package issue on your system assuming you actually followed [this](http://opentabletdriver.net/Wiki/Install/Linux#debian) install guide.

Dependencies you must have are (do keep in mind those are probably not the package names for the repository)
- libx11
- libxrandr
- libevdev2
- GTK+3
- (dotnet ofc)

Fix the dependency issues and OTD should install just fine.

jxu, 2022-05-06:
Holding broken packages means dpkg messed up somewhere so fix that first

jamesbt365, 2022-06-18:
Unclear if this is an opentabletdriver issue or not, if it isn't please say so i can close this issue if it isn't.

InfinityGhost, 2022-08-26:
Packaging issues have been resolved AFAIK. Reopen if this is still a problem.

closed 2022-08-26 by InfinityGhost