OpenTabletDriver/OpenTabletDriver #2232: Error code for open tablet driver installation

tridentlmao, 2022-05-08:
## Description

so whenever i try to open up open tablet, an error message pops up. and when i press ok or close it, open tablet closes. someone explain how to fix it
## System Information: windows 10

| Name             | Value |
| trident | ----- |
| Operating System   Windows 10      |
| OpenTabletDriver Version  Latest release|
| Tablet       xp pen g640            |
[image]

jamesbt365, 2022-06-04:
The daemon is crashing, remove the settings file in %localappdata%\OpenTabletDriver and also the plugins folder there.

jamesbt365, 2022-06-18:
Closing due to no response or extra information and from what i can see the solution is above.

closed 2022-06-18 by jamesbt365

Jupiterium, 2023-06-20:
> The daemon is crashing, remove the settings file in %localappdata%\OpenTabletDriver and also the plugins folder there.
Thanks, this helped. Had an issue with the JSON file, still not sure why it gave me an error, but thank you =)