OpenTabletDriver/OpenTabletDriver #2233: [Feature Suggestion] App specific Presets

andrey6498, 2022-05-08:
Ability to create presets with custom app binding. 
It should be something like:

Get current focus(app you currently use), including exclusive full screen apps => if focus is same as something described in presets => load corresponding preset. 

 Target App should be defined either by process name or process path.(Probably both ways are needed but they shouldn't depend on each other in interactions) 

Also it should have Global/Default preset (like if no app specific preset is present for current focus => load Global) 

This will require app to check only new processes for name and path once(we don't have to scan all processes, just imply that driver is run before anything that requires it)

X9VoiD, 2022-05-08:
If somebody knows how to retrieve "current focused window" on Linux and MacOS then we'll gladly add it.

For Linux there's also two ways it must be queried, X11 and Wayland.

vedattt, 2022-05-08:
> to maybe preload specific preset into memory

FYI, any preset on the UX is already preloaded into memory.

> Also it should have Global/Default preset (like if no app specific preset is present for current focus => load Global)

If I remember correctly, loading a preset does not overwrite `settings.json` where your "default" settings reside. In which case, it's enough to just fall back to default settings the same way we do on driver initialization currently.

Although this is a cool feature, it's unlikely to be implemented and released any time soon with various complexities including what Void mentioned. The earliest *possible* version is 0.7, which is still quite a ways off as it is a rewrite. So in the meantime, if you didn't know already: you can switch presets from the right click menu on the tray icon if you're on Windows or MacOS. It's at least faster than going into the UX every time.

andrey6498, 2022-05-10:
Some python script solution for mac(kinda questionable though but maybe worth mentioning 

[source](https://apple.stackexchange.com/questions/123730/is-there-a-way-to-detect-what-program-is-stealing-focus-on-my-mac)
```
#!/usr/bin/python from AppKit import NSWorkspace import time t = range(1,100) for i in t: time.sleep(3) activeAppName = NSWorkspace.sharedWorkspace().activeApplication()['NSApplicationName'] print activeAppName```

And i guess that `Process.GetCurrentProcess` method won't work for mac (i dont really know differences between win and mac dotnet)