OpenTabletDriver/OpenTabletDriver #2234: Cursor teleporting.

bigfootsundee, 2022-05-09:
## Description

Even after disabling raw input in osu it still teleporting.
## System Information: 

| Name             | Value |
| ---------------- | ----- |
| Operating System         | windows
| OpenTabletDriver Version | 0.6.0.4
| Tablet                   | VEIKK VK640

doubledge-spec, 2022-05-09:
Export Diagnostics and upload here.

jamesbt365, 2022-05-09:
if it is everywhere and not just osu, and file > reset to defaults doesn't fix it then you most probably have another driver installed, the otd console tab should show this.

bigfootsundee, 2022-05-09:
> 

how do i do that exactly?

jamesbt365, 2022-05-09:
As you didn't clarify what bit you were replying to i'm assuming mine.
[image]
it will say on the first few lines if you do or don't have another driver installed.

If you were after what @doubledge-spec said but didn't mention how then go `help > export diagnostics` save that file as a txt and send it here.

bigfootsundee, 2022-05-09:
> reply to james

sorry this is really complicated, give me a moment

jamesbt365, 2022-05-09:
just do the diagnostics and i'll look myself then.

bigfootsundee, 2022-05-09:
> just do the diagnostics and i'll look myself then.

i cant send txt files

jamesbt365, 2022-05-09:
Yes you can, make sure its not .txt.json. You'll know if you have file extensions on, if you can't figure that out just zip it quickly because i want to sleep

bigfootsundee, 2022-05-09:
> 

i just noticed, my bad

bigfootsundee, 2022-05-09:
> Yes you can, make sure its not .txt.json. You'll know if you have file extensions on, if you can't figure that out just zip it quickly because i want to sleep
[OTD diagnostic.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/8648097/OTD.diagnostic.zip)

jamesbt365, 2022-05-09:
Uninstall veikk drivers.
```    },
    {
      "Time": "2022-05-08T20:28:46.1865707-04:00",
      "Group": "Detect",
      "Message": "Another tablet driver found: Veikk",
      "StackTrace": null,
      "Level": 2,
      "Notification": false
    },
    {
      "Time": "2022-05-08T20:28:46.1865799-04:00",
      "Group": "Detect",
      "Message": "Detection for Veikk tablets might be impaired",
      "StackTrace": null,
      "Level": 2,
      "Notification": false
    },```

If you don't see them in the apps and features menu, install them, then uninstall them.

closed 2022-05-09 by jamesbt365

bigfootsundee, 2022-05-09:
> Uninstall veikk drivers.
>
> ```
>     {
>       "Time": "2022-05-08T20:28:46.1865707-04:00",
>       "Group": "Detect",
>       "Message": "Another tablet driver found: Veikk",
>       "StackTrace": null, [...]

hm. thought i did that already, ill try and find out how to delete it

bigfootsundee, 2022-05-09:
> Uninstall veikk drivers.
>
> ```
>     {
>       "Time": "2022-05-08T20:28:46.1865707-04:00",
>       "Group": "Detect",
>       "Message": "Another tablet driver found: Veikk",
>       "StackTrace": null, [...]
thanks for helping sleep well. ill reply if i figure it out

Sublimelime, 2022-05-09:
For the future, it's generally unnecessary/discouraged to quote-reply on github if you're replying to the most recent message, and especially if a conversation only involves two people. If you need to address someone, you can just @ them.