OpenTabletDriver/OpenTabletDriver #2242: FileNotFoundException

adrianc555, 2022-05-11:
## Description

driver starts up normally however tablet isnt detected, when clicking the detect tablet button it detects the tablet then as soon as the pen touches it will no longer detect the table and produces this line
[FileNotFoundException:Error]	Could not load file or assembly 'System.Runtime.CompilerServices.Unsafe, Version=6.0.0.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a'. The system cannot find the file specified.
   at OpenTabletDriver.Configurations.Parsers.Wacom.Intuos.IntuosTabletReport..ctor(Byte[] report)
   at OpenTabletDriver.Configurations.Parsers.Wacom.Intuos.IntuosReportParser.GetToolReport(Byte[] report)
   at OpenTabletDriver.Configurations.Parsers.Wacom.Intuos.IntuosReportParser.Parse(Byte[] report)
   at OpenTabletDriver.Devices.DeviceReader`1.Main()

*was working perfectly fine yesterday, no changes since (no windows update, or driver updates)

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         | windows 10
| OpenTabletDriver Version | 0.6.0.4
| Tablet                 | CTL-472

jamesbt365, 2022-05-11:
Could you supply a diagnostic for when this happens? Go to help > export diagnostics, save that file and send it here.

Also do file > reset to defaults, then save the settings, does it still happen then?

adrianc555, 2022-05-11:
[diagnostic.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/8669988/diagnostic.zip)
 
reset to default did nothing the issue still persists the same

> Could you supply a diagnostic for when this happens? Go to help > export diagnostics, save that file and send it here.
> 
> Also do file > reset to defaults, then save the settings, does it still happen then?

closed 2022-05-12 by adrianc555