OpenTabletDriver/OpenTabletDriver #2245: XP-Pen Star 06C massive input lag

EluWriter, 2022-05-12:
## Description
XP-Pen Star 06C experiences massive input lag. When I use my stylus, it's slow and very delayed. I installed OpenTabletDriver and launched it, changed the active area to one monitor, and that's it. Regardless of the monitor chosen or any area the lag persists.

## System Information:
CPU - AMD Ryzen 5 2600X
RAM - 16GB 3200MHz DDR4
GPU - MSI GAMING X GTX 1070
| Operating System         |
Linux Mint 20.3 Cinnamon
Cinnamon Version - 5.2.7
Linux Kernel 5.4.0-110-generic
| OpenTabletDriver Version |
v0.6.0.4
| Tablet                   |
XP-Pen Star 06C

jamesbt365, 2022-05-12:
help > export diagnostic, save that file and send it here.

EluWriter, 2022-05-12:
[diagnosticxppenstar06c.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/8676928/diagnosticxppenstar06c.txt)

jamesbt365, 2022-05-12:
Nothing seems to be getting in the way of the input. 

This can come down to a few things then:
- Something that otd can't see is blocking input.
- This is the tablets hardware smoothing, lag that opentabletdriver can't remove due to it being in the tablets firmware, thus is a hardware issue and the solution is a tablet that doesn't have hardware smoothing.
- Brain issue (your being nocebo'd)

EluWriter, 2022-05-12:
The lag is certainly not in my brain, it's very noticeable. The best analogy I can give is that it's like gaming in sub-20FPS range, maybe even lower. 
My processor is not struggling at all.
The tablet works just fine on Windows 10, but then I used the drivers from the official website. Unfortunately, I have a tablet model that is exactly one that isn't officially supported for Linux, thus OpenTabletDriver usage.

jamesbt365, 2022-05-12:
Could you get a recording then?

EluWriter, 2022-05-12:
I can record my own hand as well as the screen. I will send the video once I do that.

EluWriter, 2022-05-12:
I hope the issue is visible enough in this video. https://youtube.com/shorts/CaJEjfECCx4?feature=share

jamesbt365, 2022-05-12:
Does this happen in the tablet debugger?

go to `tablet > tablet debugger`

EluWriter, 2022-05-12:
Yes. The report rate is 5Hz and doesn't go higher, but it does drop to 3Hz and around that.

jamesbt365, 2022-05-12:
but it sounds like this http://opentabletdriver.net/Wiki/FAQ/Linux#performance-nvidia

jamesbt365, 2022-05-12:
> Yes. The report rate is 5Hz and doesn't go higher, but it does drop to 3Hz and around that.

The tablet debugger takes the data the tablet sends and displays it visually, so its strange why its doing this.

EluWriter, 2022-05-12:
I can't seem to find "Force full composition pipeline" option. The Nvidia settings I have are called "NVIDIA X Server Settings"

EluWriter, 2022-05-12:
It appears I had selective blindness of that option. I found it, and it makes no difference regardless if it's on or off.

jamesbt365, 2022-05-12:
Do you have a windows machine about?

EluWriter, 2022-05-12:
I have it in dual-boot. I can try to use OpenTabletDriver on it and see if the issue persists, I suppose.

EluWriter, 2022-05-12:
The issue is not present on Windows 10.

gonX, 2022-05-23:
If the tablet debugger report rate is truly 5Hz this sounds like an actual bug. The title of the thread should probably be 'XP-Pen Star 06C has low polling rate on Linux" or something along those lines.

jamesbt365, 2022-06-09:
@EluWriter @TimStep
If you have windows at hand:
Close OTD, install the OEM drivers then follow this:

https://desowin.org/usbpcap/thankyou.html?file=1.5.4.0/USBPcapSetup-1.5.4.0.exe Download and install this tool

- When done, head to C:\Program Files\USBPcap in the explorer (you can probably copy paste that path in the path bar)

- Once you're there, start USBPcapCMD.exe

- A console will appear, answer n at the first prompt and reboot your pc

- Head back to C:\Program  Files\USBPcap,

- Start USBPcapCMD.exe,

- Screenshot

notbot479, 2022-12-14:
Same issue with Star 06C on LMDE 5, on windows with official drivers is ok.

notbot479, 2022-12-14:
[image]

bobrity, 2023-01-28:
I have the same problem. star 06c. 5 hertz speed.

hendursaga, 2023-06-05:
Using latest stable release, on Ubuntu jammy, under Swaywm / Wayland, with an Alder Lake-P Integrated Graphics Controller, using Intel drivers, I have a rate of about 250 hertz, things work fine for me. So, perhaps we can rule out Intel?

EluWriter, 2023-06-05:
This issue was fixed for me through the official discord server support, although I've not seen any recent release that talks about it. A configuration file was made, which I then put in `/usr/share/OpenTabletDriver/Configurations`
[Star_06C.json.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/11657889/Star_06C.json.zip)

jamesbt365, 2023-06-05:
@EluWriter I would prefer if you didn't close issues without fixes properly implemented, please wait until a pull request is made before closing issues in the future.

EluWriter, 2023-06-05:
> 

Oh, sorry, didn't know about that. I'll keep it in mind for the future.

PR #2773 (2023-08-13): XP-Pen Star 06C low report rate linux fix
Verification: https://discord.com/channels/615607687467761684/723399565780189234/1101943269585326170

Fixes #2245
closed 2023-08-13 by X9VoiD