OpenTabletDriver/OpenTabletDriver #2249: Support Request: Two error messages pop up when I start OpenTabletDriver

Aiptex, 2022-05-15:
# Support Request
The first error message that pops up is this one
[image]

And then after I press ok, a second error message pops up
[image]

After I press ok on the second error message, OpenTabletDriver just closes.

## What I've already tried
I've already tried updating my .NET 6 Runtime and making sure RTSS does not hook onto OpenTabletDriver

## Diagnostic Information
I could not get the diagnostics information as OpenTabletDriver won't start up correctly.

jamesbt365, 2022-05-15:
- Open Event Viewer (you can find it by pressing the windows key and searching Event Viewer)
- Expand Windows Logs on the left menu
- Select Application
- Find an error with .NET Runtime in the Source column and check that the error is from OTD
- Copy the contents of the box where it says "Application: OpenTabletDriver..."
- Send what you have copied here

Aiptex, 2022-05-15:
Application: OpenTabletDriver.Daemon.exe
CoreCLR Version: 6.0.422.16404
.NET Version: 6.0.4
Description: The process was terminated due to an unhandled exception.
Exception Info: System.Reflection.TargetInvocationException: Exception has been thrown by the target of an invocation.
 ---> System.Collections.Generic.KeyNotFoundException: The given key 'Gaomon S620' was not present in the dictionary.
   at System.Collections.Generic.Dictionary`2.get_Item(TKey key)
   at VoiDPlugins.Library.VoiD.GlobalStore`1.Get(TabletReference tabletReference) in /home/runner/work/VoiDPlugins/VoiDPlugins/src/VoiDPlugins.Library/VoiD/GlobalStore.cs:line 45
   at VoiDPlugins.OutputMode.WindowsInkButtonHandler.Initialize(TabletReference tabletReference) in /home/runner/work/VoiDPlugins/VoiDPlugins/src/OutputMode/WindowsInk/WinInkButtonHandler.cs:line 34
   at VoiDPlugins.OutputMode.WindowsInkButtonHandler.set_Reference(TabletReference value) in /home/runner/work/VoiDPlugins/VoiDPlugins/src/OutputMode/WindowsInk/WinInkButtonHandler.cs:line 30
   --- End of inner exception stack trace ---
   at System.RuntimeMethodHandle.InvokeMethod(Object target, Span`1& arguments, Signature sig, Boolean constructor, Boolean wrapExceptions)
   at System.Reflection.RuntimeMethodInfo.InvokeOneParameter(Object obj, BindingFlags invokeAttr, Binder binder, Object parameter, CultureInfo culture)
   at System.Reflection.RuntimePropertyInfo.SetValue(Object obj, Object value, BindingFlags invokeAttr, Binder binder, Object[] index, CultureInfo culture)
   at System.Reflection.RuntimePropertyInfo.SetValue(Object obj, Object value, Object[] index)
   at OpenTabletDriver.Desktop.Reflection.PluginSettingStore.TriggerEventMethods(Object obj, TabletReference tabletReference)
   at OpenTabletDriver.Daemon.DriverDaemon.SetBindingHandlerCollectionSettings(IServiceManager serviceManager, PluginSettingStoreCollection collection, Dictionary`2 targetDict, TabletReference tabletReference)
   at OpenTabletDriver.Daemon.DriverDaemon.SetBindingHandlerSettings(InputDeviceTree dev, IOutputMode outputMode, BindingSettings settings, TabletReference tabletReference)
   at OpenTabletDriver.Daemon.DriverDaemon.SetSettings(Settings settings)
   at OpenTabletDriver.Daemon.DriverDaemon.LoadUserSettings()
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__128_1(Object state)
   at System.Threading.QueueUserWorkItemCallbackDefaultContext.Execute()
   at System.Threading.ThreadPoolWorkQueue.Dispatch()
   at System.Threading.PortableThreadPool.WorkerThread.WorkerThreadStart()
   at System.Threading.Thread.StartCallback()

jamesbt365, 2022-05-15:
Remove %localappdata%\OpenTabletDriver\Plugins\Windows ink or whatever its called

Aiptex, 2022-05-15:
Thank you! It works now

closed 2022-05-15 by Aiptex