OpenTabletDriver/OpenTabletDriver #2258: Support Request: Tablet detected but not working

Conando025, 2022-05-20:
# Support Request
OTD detects my HUION 420 but doesn't do anything with it. So no button pushes aren't registered and no cursor moment

## What I've already tried
Checked that the dotnet sdk is on the newest version.
reinstalled the driver

looked at the errors in the console but there are no github issues with the same error mentioned (Unable to retrieve device string)

## Diagnostic Information
[Diagnostics.json.log](https://github.com/OpenTabletDriver/OpenTabletDriver/files/8741739/Diagnostics.json.log)

jamesbt365, 2022-05-20:
try this

echo "blacklist hid_uclogic" | sudo tee -a /etc/modprobe.d/blacklist.conf
sudo rmmod hid_uclogic

Conando025, 2022-05-20:
Did not help. For the second command I got a `rmmod: ERROR: Module hid_uclogic is not currently loaded` back

jamesbt365, 2022-05-20:
are you sure your pen isn't dead?

Conando025, 2022-05-20:
Well the tablet notices the pen since the indicator light goes on and additionally before installing the otd it somewhat worked so yeah

jamesbt365, 2022-05-20:
tablet > device string reader
9580 top box
110 middle
dump all
save that file, send it here

Conando025, 2022-05-20:
Ok we're getting somewhere, it just plops up the the box telling to `Please replug the tablet, and then press OK to continue` but doing so doesn't help but the client in the background detects, that a HUION 420 is plugged int

Conando025, 2022-05-20:
Ok after dumping a few values per hand it allowed me to dump all, weird
[Huion.420.log](https://github.com/OpenTabletDriver/OpenTabletDriver/files/8743914/Huion.420.log)

Conando025, 2022-05-20:
Oh wow I got to resolve it by switching the cable. I'm still a bit confused as to how ended up with this set of issues but with that my issue is solved

closed 2022-05-20 by Conando025