OpenTabletDriver/OpenTabletDriver #2259: Auxiliary Buttons configuration not in GUI for HUION 420

Conando025, 2022-05-20:
## Description
For some reason the 3 Auxiliary Buttons on the HUION 420 are not configurable through the GUI
Configurations done through the settings json are working

## System Information:
| Name             | Value |
| ---------------- | ----- |
| Operating System         | Linux (Pop_os!) |
| OpenTabletDriver Version | 0.6.0.4 |
| Tablet                   | HUION 420 |

jamesbt365, 2022-05-20:
The Huion 420 doesn't have buttons...

Conando025, 2022-05-20:
Oh so then its not detecting right damn

jamesbt365, 2022-05-20:
Nope, blame the firmware engineers at huion, they fucked it up like they do fuck things up, i'll send a config with buttons on but its huions fault

jamesbt365, 2022-05-20:
- Make a folder in `/usr/share/OpenTabletDriver` called `Configurations`
- unzip and move the .json there. [420.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/8744150/420.zip)
- Replug your tablet.

closed 2022-05-20 by jamesbt365

Conando025, 2022-05-20:
Oh wow thanks for the quick response, now that i know what the issue is i found the configuration options sorry to bother you again