OpenTabletDriver/OpenTabletDriver #2261: Updater Error

M1Ck3yCS, 2022-05-22:
When im installing the update with the built in updater, it said "Error: RemoteInvocationException 
## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         | Window 10
| OpenTabletDriver Version | 0.6.0.3
| Tablet                   | CTL 480
[image]
[image]

joshdee, 2022-05-22:
Got that same bug too, it's annoying.

[Windows 11, Gaomon S620]
[image]

jamesbt365, 2022-05-22:
Update manually, if it still doesn't work delete `%localappdata%\OpenTabletDriver` make sure all instances of otd are dead then start it again.

GabrielSJones, 2022-07-25:
James' replay about deleting appdata\otd folder didnt work for me, here's how I was able to get it to work.
1) dl zip of newest install and extract
2) close daemon in task manager (daemon still runs even if you close gui application)
2.5) here i also deleted otd folders in appdata and appdata\temp
3) run new daemon exe first, and it will tell you if it installs properly
4) now running the regular gui app should work and not prompt updates

X9VoiD, 2022-12-31:
OTD.Updater is no more. F

closed 2022-12-31 by X9VoiD