OpenTabletDriver/OpenTabletDriver #2262: Xencelabs Pen Tablet Medium is not recognized

y-usuzumi, 2022-05-22:
## Description
My Tablet comes with Xencelabs Pen Tablet Medium Bundle SE. However OTD doesn't recognize it:

[image]

In the output of `lsusb`:

```
Bus 003 Device 002: ID 28bd:5201 XP-Pen Xencelabs Pen Tablet Medium
```

The website claims it is the same model as the black version according to their website (BPH1212W-K02-A). However in Device String Reader it appears to be a different model:

[image]

When I use the Configuration Editor to dump the configuration, I got the following errors:

[image]

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System         |Arch Linux|
| OpenTabletDriver Version |v0.6.0.4+f4c255a9
| Tablet                   |Xencelabs Pen Tablet Medium|

jamesbt365, 2022-05-22:
help > export diagnostic, save that file, send it here

y-usuzumi, 2022-05-22:
[diagnostics.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/8750441/diagnostics.txt)

Thanks!

jamesbt365, 2022-05-22:
[Pen Tablet Medium.json.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/8750444/Pen.Tablet.Medium.json.txt)

remove the txt from this files name, move it to `/usr/share/OpenTabletDriver/Configurations`

y-usuzumi, 2022-05-22:
Awesome! That fixed the problem.

Thanks a lot!

jamesbt365, 2022-05-22:
@y-usuzumi get a diagnostic with the tablet detected and working please

y-usuzumi, 2022-05-22:
There you go!
[diagnostic_working.json.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/8750473/diagnostic_working.json.txt)

Side note: the MaxX and MaxY values of (52324.0, 29600.0) doesn't work well for my tablet. I could reach only about 3/4 of the entire screen horizontally. Not sure if it's a model difference but in the "Tablet debugger" window the max values are (32768, 32768). After updating the values it worked well.

jamesbt365, 2022-05-22:
Tablet isn't in fully working state yet.

jamesbt365, 2022-05-22:
get a full string  dump as well, as you did before but a full dump, send that here

y-usuzumi, 2022-05-22:
Did you mean in the Device String Reader window? I'm stuck here:

[image]

Replugging doesn't help.

jamesbt365, 2022-05-22:
what did you input into the boxes?

y-usuzumi, 2022-05-22:
First nothing, then 10429, 20993, 1, and then 10429, 20993, empty. None of them worked

jamesbt365, 2022-05-22:
do you have a windows dual boot or a virtual machine you can pass usbs through?

y-usuzumi, 2022-05-22:
I have Windows dual boot and a Ubuntu 22.04 VM that I created yesterday.

Please let me know what I can do to help. I need to head out now. I'll be able to help when I'm back. Thanks!

y-usuzumi, 2022-05-23:
Hi, anything I can do?

jamesbt365, 2022-05-23:
on Windows, Close OTD, install the OEM drivers then follow this:

https://desowin.org/usbpcap/thankyou.html?file=1.5.4.0/USBPcapSetup-1.5.4.0.exe Download and install this tool

- When done, head to C:\Program Files\USBPcap in the explorer (you can probably copy paste that path in the path bar)

- Once you're there, start USBPcapCMD.exe

- A console will appear, answer n at the first prompt and reboot your pc

- Head back to C:\Program  Files\USBPcap,

- Start USBPcapCMD.exe,

- Screenshot

y-usuzumi, 2022-05-24:
[image]

I forgot to connect my tablet before I ran it for the first time. Then I plugged it in and restarted and it is still not there. Is that expected?

X9VoiD, 2022-05-24:
This is expected, it's HID-compliant pen.

y-usuzumi, 2022-05-25:
@X9VoiD Thanks. Yes it was the only tablet connected.

jamesbt365, 2022-05-25:
1) press 2 then press enter,

2) enter a name such as capture then press enter,

3) Do the following steps:

- Move for 5s
- Unplug the tablet from your computer (remove the usb cable from the port),
- Wait 2 seconds,
- Plug in your tablet again (connect your tablet to your pc ),
- Wait 2 seconds,
- Move for 5s
- Close the console (red cross),
- A new file has been generated with the name you gave in the same folder,
- Upload that file here

y-usuzumi, 2022-05-25:
[capture.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/8768714/capture.zip)

InfinityGhost, 2022-08-26:
Issue is stale, reopen if the problem still exists.

closed 2022-08-26 by InfinityGhost