OpenTabletDriver/OpenTabletDriver #2266: Support Request:

zzJoshS, 2022-05-25:
# Support Request
Hello I've been having issues with my cpu usage when using OpenTabletDriver.

It's always at minimum 28 percent usage 

[image]
[image]

My CPU is an intel i3-10100f
operating system is Windows 10
It expects a high CPU usage because of an fallback timer but I don't know what that means.

## What I've already tried
I didn't find any other issues that are similiar to my situation.
Or I couldn't tell if someone had the same problem.
I also downloaded old releases which luckily worked but after v.6.0 it started again to increase.
My filter combination would always begin to jitter which stops only on v.6.0 and above

## Diagnostic Information
[25.05.2022.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/8773593/25.05.2022.zip)

zzJoshS, 2022-05-25:
Dumb question but how do I change the frequencies

jamesbt365, 2022-05-25:
will be a thing for every interpolator you are using, this is just an example with devocub