OpenTabletDriver/OpenTabletDriver #2278: Add "Minimum OTD Version" to TABLETS.md

X9VoiD, 2022-06-01:
## Description

This will help with avoiding confusion on why some tablets labeled as "Supported" is not actually supported on latest releases.

Tablets that are newly added will have a version labeled as `Next Release`. On release, these versions should be changed to whatever the release version is. For the initial commit, we may just seed minimum OTD version from the latest release, and then check what new configs `master` has and label them `Next Release`.

Since releases are automated via CI, the replacement from `Next Release` to `<release>` can also be automated via a script.

gonX, 2023-10-03:
> Since releases are automated via CI, the replacement from Next Release to <release> can also be automated via a script.

Can CI commit to the repository? If not, this seems like something that must be run by hand.

jamesbt365, 2023-10-03:
Also how do we plan on handling configurations which get new identifiers in the next release?