OpenTabletDriver/OpenTabletDriver #2281: [macos]window disappears when tabbing

sealvessel, 2022-06-02:
## Description
clicking the other tab disappears window until quitting

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System      macos   |
| OpenTabletDriver Version | latest
| Tablet         

https://user-images.githubusercontent.com/66351949/171587489-0ac086c1-057d-4411-9f85-28ff847b007d.mov

          |

InfinityGhost, 2022-06-02:
Don't put "latest" as the version in the template, provide the number in the titlebar or in the about window.

We've seen people put "latest" before and they were using a 2 year old version of OpenTabletDriver.

sealvessel, 2022-06-04:
> Don't put "latest" as the version in the template, provide the number in the titlebar or in the about window.
> 
> We've seen people put "latest" before and they were using a 2 year old version of OpenTabletDriver.

I updated now.

sealvessel, 2022-06-04:
Also, how do I add tags. I can't find an option to add macOS tags.

gonX, 2025-01-24:
Does this still happen on v0.6.5.0?

gonX, 2026-10-02:
I'm going to close this as assumed resolved at some point during 0.6's development cycle

closed 2026-10-02 by gonX