OpenTabletDriver/OpenTabletDriver #2283: Huion Kamvas Pro 13 on Manjaro Linux: touchbar/scrollbar not assignable in UI

wwklnd, 2022-06-02:
I wasn't sure if this should be marked as a bug, a support request, or a feature request, so I went with support request. I tried looking through both open and closed issues to see if anyone else has reported this, but I couldn't really find anything. Might have missed something, though! In that case I apologise.

I just switched to OTD from a hacked-together python script, and am really liking it so far! It seems to be a much smoother experience than anything else I've used to get my Kamvas 13 Pro working on Linux. However, I noticed that I am not able to use the touchbar/scrollbar (marked in the picture below), and it can't be assigned in the OTD GUI under Pen Settings or Auxiliary Settings.

[image]

To be clear, I couldn't get this to work with any other driver either, but having it working would be a huge improvement. Also, when I use the Tablet debugger, it _does_ show input being registered when the scrollbar is used, as does `usbhid-dump`.

## What I've already tried
I've gone through the [Linux FAQ](https://opentabletdriver.net/Wiki/FAQ/Linux) without finding anything that works, though for some reason I have to run `sudo rmmod hid_uclogic` to get OTD to register my tablet at all if I unplug it and replug it, run `usbhid-dump` or anything else that gets in the way. I tried both blacklisting `hid_uclogic` and rebuilding initramfs, but neither solved that.

## Diagnostic Information
Here's my exported diagnostics (github didn't let me upload the json file so I made it a txt instead):
[OTDiagnostics.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/8826485/OTDiagnostics.txt)

Thanks in advance for any help!

jamesbt365, 2022-06-02:
Looks like the TABLETS.md is wrong for that tablet then, scrollbars are globally not supported on OpenTabletDriver at the moment in time, but they are in the works. Sorry for the confusion.

wwklnd, 2022-06-02:
> Looks like the TABLETS.md is wrong for that tablet then, scrollbars are globally not supported on OpenTabletDriver at the moment in time, but they are in the works. Sorry for the confusion.

Ah, thanks for the quick response! Then it makes more sense. I appreciate it! :smiley:

closed 2022-06-02 by wwklnd