OpenTabletDriver/OpenTabletDriver #2288: Support Request: Configuring XP-PEN Deco mini7 using GUI

Lokawn, 2022-06-03:
# Support Request
 
- I have set the express key bindings for my tablet using the GUI but multi key bindings don't work neither in the DE nor in Xournalpp, single key binding works.
- I searched on your wiki and GitHub but was not able to find any info regarding what exactly does artist mode does, I think I can't bind my stylus buttons to mouse key in this mode. Is there any other difference from Absolute mode?
- When I was using wacom driver, I used to set area of tablet to three times its size as it provided me with the best writing experience, but it does not work in OTD the width and height fields reset themselves every time I open the app and I don't think it is working.
- The debugger shows nine express keys on the tablet instead of 8.

[image]

## Diagnostic Information

[Diagnostics](https://github.com/OpenTabletDriver/OpenTabletDriver/files/8830874/diagnostics.txt)
[Settings](https://github.com/OpenTabletDriver/OpenTabletDriver/files/8830878/settings.txt)

### I am including my system info if any of this turns out to be a bug.

| Name             | Value |
| ---------------- | ----- |
| Operating System| Arch Linux  5.18.1-arch1-1 x86_64        |
| OpenTabletDriver Version | v0.6.0.3 |
| Tablet                   | XP-PEN Deco mini7 |

jamesbt365, 2022-06-09:
>I searched on your wiki and GitHub but was not able to find any info regarding what exactly does artist mode does, I think I can't bind my stylus buttons to mouse key in this mode. Is there any other difference from Absolute mode?

artist mode is absolute mode with pressure, and its on the website.

>The debugger shows nine express keys on the tablet instead of 8.

The debugger shows how many maximum buttons is in the parser, which in this case is 9.

I'm not quite sure why you made an issue for this, what is the problem?

jamesbt365, 2022-06-18:
Closing as i think i have cleared everything up and no response

closed 2022-06-18 by jamesbt365