OpenTabletDriver/OpenTabletDriver #2289: Menu bar application MacOS

sealvessel, 2022-06-04:
## Description

Allow it to run only in the menu bar. Have an option where it is hidden from the dock and only SHOWS in the menu bar. 
[image]
Currently, I have to have it open in the dock.

jamesbt365, 2022-06-04:
Currently a limitation with opentabletdriver's ui not being able to have differences between platforms meaning that a toggle can't really be done without the ui being able to differentiate between platforms.

poof86, 2025-01-09:
I recommend hiding the dock icon by default
https://github.com/OpenTabletDriver/OpenTabletDriver/issues/3008#issuecomment-2580492004

pichfl, 2026-01-15:
Note that the situation is slightly more complex than just adding the two lines mentioned in #3008 

Setting `LSUIElement` marks the app as a background agent. This mode prevents the app from presenting a menu bar, so we need to provide everything in the menu controlled by the tray icon. 

Is this being worked on? Otherwise I'd love to give this a try.