OpenTabletDriver/OpenTabletDriver #2291: Can't double click even with antichatter

sealvessel, 2022-06-04:
## Description
I first tried to click with mouse. It works. 
I then secondly tried to click with mouse while hovering over with pen. It works. 
Then I click myself with tablet pen on tablet, then tried clicking with on-pen button. Does not work. 

Since I tried to click while hovering and it works while steady, then it means that there is something wrong with the left click passing information to operating system. 

## System Information:

| Name             | Value |
| ---------------- | ----- |
| Operating System | MacOS 12.4
| OpenTabletDriver Version | 0.6.0.4
| Tablet | Wacom CTL-472

https://user-images.githubusercontent.com/66351949/171984866-b8c160fe-1383-4c46-9427-4ab1bc485f76.mov

X9VoiD, 2022-06-04:
OTD uses MacOS's own APIs for sending left clicks, as seen from [here](https://github.com/OpenTabletDriver/OpenTabletDriver/blob/437f1c25fcc2b24f7846e00615851667a75c044d/OpenTabletDriver.Native/OSX/Input/CGMouseButton.cs#L5) for declaring what to send and [here](https://github.com/OpenTabletDriver/OpenTabletDriver/blob/437f1c25fcc2b24f7846e00615851667a75c044d/OpenTabletDriver.Desktop/Interop/Input/MacOSVirtualMouse.cs#L154) for how it's sent. There are no further variations for this code, and as such the problem lies on either the user clicking too slowly or the operating system.

sealvessel, 2022-06-05:
> OTD uses MacOS's own APIs for sending left clicks, as seen from [...]

I have tried kuuube antichatter at option 200, still does not work with tablet clicking. 

https://user-images.githubusercontent.com/66351949/172045531-3d3625ae-2097-485e-b9d8-4bf87495cefd.mov

mateowatson, 2022-08-24:
I have the same issue I believe. I cannot double click on anything. It just doesn't register as a double click even when double click speed is set all the way low.

jamesbt365, 2022-08-24:
because macos has to have a separate thing (because its stupid) for double clicking.

woutervanwijk, 2022-09-13:
So double clicking on MacOS doesn't work at all? (For me indeed not)

jamesbt365, 2022-09-13:
Yes, as you can see above that is the case.

jamesbt365, 2023-03-22:
Platform specific options are not currently on the menu, sorry. You can use script runner and an application in the meantime that can send a double click.

subfighter3, 2023-03-22:
I see, thanks for the quick reply @jamesbt365 
Do you know if the such application exists already? I don't know how to develop a proper script to do that, but if there is an app able to emulate this behaviour. Also, the pan/scroll feature is something really needed, and if it would be possible to emulate with some external app to use with the script runner it would be great

closed 2024-11-25 by InfinityGhost

subfighter3, 2024-11-25:
Opening back this issue because it is not completed at all!
Can somebody post here the necessary scripts to enable double click and possibly the pan feature? Thanks in advance

jamesbt365, 2024-11-25:
Did you test on the latest 0.6.x branch (which is NOT the latest release)

subfighter3, 2024-11-25:
not really, but I can't understand from where to download that version for MacOS. 
From Releases I can only download the 0.6.4.0 from the Master branch…

subfighter3, 2024-11-25:
Yep this is what I did to swap branch, but the Releases link on the right sidebar still refers to the master branch, so how can I download a release from the 0.6.x branch?

jamesbt365, 2024-11-25:
> Yep this is what I did to swap branch, but the Releases link on the right sidebar still refers to the master branch, so how can I download a release from the 0.6.x branch?

The *releases* section shows *releases* of the driver, they do not update per commit on a branch and there are not different releases per branch. As 0.6.5.0 has not been released, theres no release for the commits past 0.6.4.0 available from 0.6.x yet.

Either download an [action](https://github.com/OpenTabletDriver/OpenTabletDriver/actions?query=branch%3A0.6.x) or compile it yourself.

subfighter3, 2024-11-25:
Ok thanks for your reply, I see what you mean, sadly I don't know how to compile nor run an action in order to get an executable app :( can somebody run the action and just link here the app?

@jamesbt365 thanks anyway for pointing that out :)

jamesbt365, 2024-11-25:
> Ok thanks for your reply, I see what you mean, sadly I don't know how to compile nor run an action in order to get an executable app :( can somebody run the action and just link here the app?

I linked you precompiled actions, you just need to click on one.

subfighter3, 2024-11-25:
Thanks, but nope, I tried some and I can't find builds, just source files. Some older actions has something that looks like builds but they are expired and not downloadable, but also 1 year old so prolly not what I'm looking for

jamesbt365, 2024-11-25:
Please come down to [our discord server](https://discord.gg/9bcMaPkVAR) if you want extra guidance, support via GitHub is slow and we're just repeatedly hitting peoples inboxes over something that has already been fixed at this point.

If you don't have a Discord account or do not want to join our Discord server, wait until the next release of OpenTabletDriver that will be releasing over the coming weeks.