OpenTabletDriver/OpenTabletDriver #2296: cannot bind the pen on the GUI

g1bus, 2022-06-07:
OS: Endeavor os
Tablet: Wacom CTL-472 

Hey,
I just go the Tablet driver working today, and its great. I just cannot bind the buttons, or the tip of the pen. it appears that i have set the binds, But in testing, it has not varied from the default settings.
I just want to clarify if I'm missing something, or if there is a actual fix for this.
everything else on the GUI works flawlessly.
thanks

jamesbt365, 2022-06-07:
Fill out the template properly.

jamesbt365, 2022-06-07:
From your edit i can see that you have a 100% threshold on, this means you have to push the pen **ALL** of the way down for it to activate the binding. Attach a diagnostic as shown in the template or i will not be able to help you further.

sealvessel, 2022-06-10:
put it on 99% and it may work

jamesbt365, 2022-06-18:
Closing as it entirely looks like user error and the lack of providing a diag.

closed 2022-06-18 by jamesbt365