OpenTabletDriver/OpenTabletDriver #2302: Spamming side buttons causes crash on macOS, also persists dragged element even when quit.

sealvessel, 2022-06-10:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Steps to Reproduce

1. Bind a button to right click on the side button. 
2. Repeatedly hover and click it, and hover, press side button, then unhover while holding down the button, rinse and repeat. 
3. Wait until it crashes on macOS. 

### Expected Behavior

Nothing should crash

### Observed Behavior

Crashes when I use side buttons repeatedly

### Tablet Device

Wacom CTL-472

### Diagnostic Information

{
  "App Version": "OpenTabletDriver v0.6.0.4", 
  "Build Date": "05/04/2022", 
  "Operating System": {
    "Platform": 4, 
    "ServicePack": "", 
    "Version": "12.4.0", 
    "VersionString": "Unix 12.4.0"
  }, 
  "Environment Variables": {
    "USER": "o", 
    "TEMP": null, 
    "TMP": null, 
    "TMPDIR": "/var/folders/x0/yks1s3j96gb1t2wk90lvm4t40000gn/T/"
  }, 
  "HID Devices": [
    {
      "DevicePath": "IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC1@14/XHC1@14000000/HS14@14700000/CTL-472@14700000/IOUSBHostInterface@1/AppleUserUSBHostHIDDevice", 
      "Manufacturer": "Wacom Co.,Ltd.", 
      "ProductName": "CTL-472", 
      "SerialNumber": "1KA00L1040045", 
      "FriendlyName": "CTL-472", 
      "VendorID": 1386, 
      "ProductID": 890, 
      "InputReportLength": 64, 
      "OutputReportLength": 0, 
      "FeatureReportLength": 0, 
      "CanOpen": true
    }, 
    {
      "DevicePath": "IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC1@14/XHC1@14000000/HS08@14300000/USB Receiver@14300000/IOUSBHostInterface@2/AppleUserUSBHostHIDDevice", 
      "Manufacturer": "Logitech", 
      "ProductName": "USB Receiver", 
      "SerialNumber": "", 
      "FriendlyName": "USB Receiver", 
      "VendorID": 1133, 
      "ProductID": 50489, 
      "InputReportLength": 32, 
      "OutputReportLength": 32, 
      "FeatureReportLength": 0, 
      "CanOpen": true
    }, 
    {
[... 2069 lines cut ...]
      "Time": "2022-06-10T15:20:44.595079+08:00", 
      "Group": "Wacom CTL-472", 
      "Message": "Tip Binding: [Mouse Button Binding: Left]@0%", 
      "StackTrace": null, 
      "Level": 1, 
      "Notification": false
    }, 
    {
      "Time": "2022-06-10T15:20:44.595084+08:00", 
      "Group": "Wacom CTL-472", 
      "Message": "Pen Bindings: Mouse Button Binding: Right, Mouse Button Binding: Left", 
      "StackTrace": null, 
      "Level": 1, 
      "Notification": false
    }, 
    {
      "Time": "2022-06-10T15:20:44.595089+08:00", 
      "Group": "Settings", 
      "Message": "Driver is enabled.", 
      "StackTrace": null, 
      "Level": 1, 
      "Notification": false
    }
  ]
}

jamesbt365, 2022-06-10:
please provide a diagnostic as an attachment as a txt file and don't just paste it in, it makes it hard to navigate (in this case diagnostics won't be necessary but its just hard to get around the issue)

gonX, 2025-12-05:
Does this still happen on more recent versions of OpenTabletDriver?