OpenTabletDriver/OpenTabletDriver #2305: Support Request:

sb-child, 2022-06-11:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### What do you need help with?

OS: NixOS 22.05
`otd-gui` Version: 0.6.0.4
`otd-daemon` Version: 0.6.0.4

OpenTabletDriver can't detect `Gaomon 1060 Pro`
after trying to solve permission issues, i can't see any error but no tablets were detected:
[image]
[image]

`lsusb` output:
```
$ lsusb
Bus 003 Device 002: ID 256c:0064 Gaomon Gaomon Tablet_1060Pro
```

### What have you tried?

i tried to restart my computer, and add user-define udev on `configuration.nix`:
```
  services.udev = {
    # Gaomon 1060 Pro
    extraRules = ''
      SUBSYSTEM=="hidraw", ATTRS{idVendor}=="256c", ATTRS{idProduct}=="006d", MODE="0666"
      SUBSYSTEM=="hidraw", ATTRS{idVendor}=="256c", ATTRS{idProduct}=="0064", MODE="0666"
      SUBSYSTEM=="usb", ATTRS{idVendor}=="256c", ATTRS{idProduct}=="006d", MODE="0666"
      SUBSYSTEM=="usb", ATTRS{idVendor}=="256c", ATTRS{idProduct}=="0064", MODE="0666"
      SUBSYSTEM=="input", ATTRS{idVendor}=="256c", ATTRS{idProduct}=="006d", ENV{LIBINPUT_IGNORE_DEVICE}="1"
      SUBSYSTEM=="input", ATTRS{idVendor}=="256c", ATTRS{idProduct}=="0064", ENV{LIBINPUT_IGNORE_DEVICE}="1"
    '';
  };
```

### Diagnostic Information

but it crashed:
```
(OpenTabletDriver.UX.Gtk:9375): GLib-GIO-ERROR **: 23:50:12.547: No GSettings schemas are installed on the system
[1]    9375 IOT instruction (core dumped)  otd-gui
```

jamesbt365, 2022-06-11:
`otd getdiagnostics > diag.txt`

sb-child, 2022-06-11:
> `otd getdiagnostics > diag.txt`
[diag.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/8884278/diag.txt)

jamesbt365, 2022-06-11:
otd getstring 9580 100 200

sb-child, 2022-06-11:
> otd getstring 9580 100 200

[image]
maybe need hexdump
[image]

jamesbt365, 2022-06-11:
instead of 200 do 201, my mistake

sb-child, 2022-06-11:
> 

```
$ otd getstring 9580 100 201
GM001_T213_210615
```

jamesbt365, 2022-06-11:
This is indeed a different variant, let me make a configuration real quick.

jamesbt365, 2022-06-11:
[1060 Pro.json.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/8884302/1060.Pro.json.txt)
put this in `~/.local/share/OpenTabletDriver/Configurations`

jamesbt365, 2022-06-11:
make sure to remove the .txt at the end

sb-child, 2022-06-11:
> [1060 Pro.json.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/8884302/1060.Pro.json.txt) put this in `~/.local/share/OpenTabletDriver/Configurations`

seems this file is not loaded. 
and i modified this file
[image]
but seems not found on [diag.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/8884334/diag.txt)

maybe other way to load this configures?

jamesbt365, 2022-06-11:
restart opentabletdriver

sb-child, 2022-06-11:
> restart opentabletdriver

thank you. i found
```
ExecStart=/nix/store/4kv8y4di2pw705c6qijfq3zf2fq4a3ji-OpenTabletDriver-0.6.0.4/bin/otd-daemon -c /nix/store/4kv8y4di2pw705c6qijfq3zf2fq4a3ji-OpenTabletDriver-0.6.0.4/lib/OpenTabletDriver/Configurations
```
on `opentabletdriver.service`, so user-define config is not loaded.

i am running daemon manually and it is working
```
$ otd-daemon
...
[Device:Debug]  Initializing device 'Gaomon Tablet_1060Pro' /sys/devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2:1.0/0003:256C:0064.0004/hidraw/hidraw1
[Device:Debug]  Using report parser type 'OpenTabletDriver.Configurations.Parsers.UCLogic.UCLogicReportParser'
[Device:Debug]  Initialized string index 200
[Detect:Info]   Found tablet 'Gaomon 1060 Pro update'
[Evdev:Debug]   Successfully initialized virtual tablet. (code NONE)
[Gaomon 1060 Pro update:Info]   Output mode: Absolute Mode
[Gaomon 1060 Pro update:Info]   Display area: [1920x1080@<960, 540>:0°],
[Gaomon 1060 Pro update:Info]   Tablet area: [254x158.75@<127, 79.375>:0°],
[Gaomon 1060 Pro update:Info]   Clipping: Enabled
[Gaomon 1060 Pro update:Info]   Ignoring reports outside area: Disabled
[Gaomon 1060 Pro update:Info]   Tip Binding: [Mouse Button Binding: Left]@0%
[Settings:Info] Driver is enabled.
```

closed 2022-06-11 by sb-child

jamesbt365, 2022-06-11:
as the service is ran as the user and otd is told to check all locations it shouldn't have a problem with loading configs in the user directory.

Anyway fixed with #2239