OpenTabletDriver/OpenTabletDriver #2310: OpenTabletDriver.Daemon: ExtendedSocketException on start

SimonWoolf, 2022-06-14:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Steps to Reproduce

Daemon crashes on start. Not sure how to reproduce on anyone else's computer; on my computer it happens every time

### Expected Behavior

-

### Observed Behavior

-

### Tablet Device

Wacom CTL-472

### Diagnostic Information

Unable to get diagnostics as the daemon won't start.
Syslog has the stacktrace:
```
Jun 14 20:49:47 simon-thinkpad systemd[2649]: Started OpenTabletDriver Daemon.
Jun 14 20:49:47 simon-thinkpad OpenTabletDriver.Daemon[1043545]: [Detect:Info]#011Searching for tablets...
Jun 14 20:49:47 simon-thinkpad OpenTabletDriver.Daemon[1043545]: [Detect:Debug]#011Searching for tablet '10moon 1060N'
... [snip lots of other devices]
Jun 14 20:49:48 simon-thinkpad OpenTabletDriver.Daemon[1043545]: [Device:Debug]#011Initializing device 'CTL-472' /sys/devices/pci0000:00/0000:00:14.0/usb3/3-7/3-7:1.0/0003:056A:037A.0016/hidraw/hidraw1
Jun 14 20:49:48 simon-thinkpad OpenTabletDriver.Daemon[1043545]: [Device:Debug]#011Using report parser type 'OpenTabletDriver.Configurations.Parsers.Wacom.Intuos.IntuosReportParser'
Jun 14 20:49:48 simon-thinkpad OpenTabletDriver.Daemon[1043545]: [Device:Debug]#011Set device feature: 02-02
Jun 14 20:49:48 simon-thinkpad OpenTabletDriver.Daemon[1043545]: [Detect:Info]#011Found tablet 'Wacom CTL-472'
Jun 14 20:49:48 simon-thinkpad OpenTabletDriver.Daemon[1043545]: [Device:Debug]#011Initializing device 'CTL-472' /sys/devices/pci0000:00/0000:00:14.0/usb3/3-7/3-7:1.1/0003:056A:037A.0017/hidraw/hidraw2
Jun 14 20:49:48 simon-thinkpad OpenTabletDriver.Daemon[1043545]: [Device:Debug]#011Using report parser type 'OpenTabletDriver.Configurations.Parsers.Wacom.Wacom64bAuxReportParser'
... [snip lots of other devices]
Jun 14 20:49:48 simon-thinkpad OpenTabletDriver.Daemon[1043545]: [Detect:Debug]#011Searching for tablet 'XP-Pen Star G960S'
Jun 14 20:49:48 simon-thinkpad OpenTabletDriver.Daemon[1043545]: Unhandled exception. System.Net.Internals.SocketExceptionFactory+ExtendedSocketException (99): Cannot assign requested address /run/user/1000/:0
Jun 14 20:49:48 simon-thinkpad OpenTabletDriver.Daemon[1043545]:    at System.Net.Sockets.Socket.DoConnect(EndPoint endPointSnapshot, SocketAddress socketAddress)
Jun 14 20:49:48 simon-thinkpad OpenTabletDriver.Daemon[1043545]:    at System.Net.Sockets.Socket.Connect(EndPoint remoteEP)
Jun 14 20:49:48 simon-thinkpad OpenTabletDriver.Daemon[1043545]:    at WaylandNET.Client.WaylandClientConnection.ConnectToDisplay(String display)
Jun 14 20:49:48 simon-thinkpad OpenTabletDriver.Daemon[1043545]:    at WaylandNET.Client.WaylandClientConnection..ctor(String display)
Jun 14 20:49:48 simon-thinkpad OpenTabletDriver.Daemon[1043545]:    at OpenTabletDriver.Desktop.Interop.Display.WaylandDisplay..ctor()
Jun 14 20:49:48 simon-thinkpad OpenTabletDriver.Daemon[1043545]:    at OpenTabletDriver.Desktop.Interop.DesktopInterop.ConstructLinuxDisplay()
Jun 14 20:49:48 simon-thinkpad OpenTabletDriver.Daemon[1043545]:    at OpenTabletDriver.Desktop.Interop.DesktopInterop.get_VirtualScreen()
Jun 14 20:49:48 simon-thinkpad OpenTabletDriver.Daemon[1043545]:    at OpenTabletDriver.Desktop.Reflection.DesktopPluginManager.<>c.<ResetServices>b__30_5()
Jun 14 20:49:48 simon-thinkpad OpenTabletDriver.Daemon[1043545]:    at OpenTabletDriver.Desktop.Reflection.ServiceManager.GetService(Type serviceType)
Jun 14 20:49:48 simon-thinkpad OpenTabletDriver.Daemon[1043545]:    at OpenTabletDriver.Desktop.Reflection.ServiceManager.GetService[T]()
Jun 14 20:49:48 simon-thinkpad OpenTabletDriver.Daemon[1043545]:    at OpenTabletDriver.Desktop.Profiles.AbsoluteModeSettings.GetDefaults(DigitizerSpecifications digitizer)
Jun 14 20:49:48 simon-thinkpad OpenTabletDriver.Daemon[1043545]:    at OpenTabletDriver.Desktop.Profiles.Profile.GetDefaults(TabletReference tablet)
Jun 14 20:49:48 simon-thinkpad OpenTabletDriver.Daemon[1043545]:    at OpenTabletDriver.Desktop.Profiles.ProfileCollection.<>c.<.ctor>b__2_0(TabletReference s)
Jun 14 20:49:48 simon-thinkpad OpenTabletDriver.Daemon[1043545]:    at System.Linq.Utilities.<>c__DisplayClass2_0`3.<CombineSelectors>b__0(TSource x)
Jun 14 20:49:48 simon-thinkpad OpenTabletDriver.Daemon[1043545]:    at System.Linq.Enumerable.SelectIListIterator`2.MoveNext()
Jun 14 20:49:48 simon-thinkpad OpenTabletDriver.Daemon[1043545]:    at System.Collections.Generic.List`1..ctor(IEnumerable`1 collection)
Jun 14 20:49:48 simon-thinkpad OpenTabletDriver.Daemon[1043545]:    at System.Collections.ObjectModel.ObservableCollection`1.CreateCopy(IEnumerable`1 collection, String paramName)
Jun 14 20:49:48 simon-thinkpad OpenTabletDriver.Daemon[1043545]:    at System.Collections.ObjectModel.ObservableCollection`1..ctor(IEnumerable`1 collection)
Jun 14 20:49:48 simon-thinkpad OpenTabletDriver.Daemon[1043545]:    at OpenTabletDriver.Desktop.Profiles.ProfileCollection..ctor(IEnumerable`1 profiles)
Jun 14 20:49:48 simon-thinkpad OpenTabletDriver.Daemon[1043545]:    at OpenTabletDriver.Desktop.Profiles.ProfileCollection..ctor(IEnumerable`1 tablets)
Jun 14 20:49:48 simon-thinkpad OpenTabletDriver.Daemon[1043545]:    at OpenTabletDriver.Desktop.Settings.GetDefaultProfiles()
Jun 14 20:49:48 simon-thinkpad OpenTabletDriver.Daemon[1043545]:    at OpenTabletDriver.Desktop.Settings.GetDefaults()
Jun 14 20:49:48 simon-thinkpad OpenTabletDriver.Daemon[1043545]:    at OpenTabletDriver.Daemon.DriverDaemon.ResetSettings()
Jun 14 20:49:48 simon-thinkpad OpenTabletDriver.Daemon[1043545]:    at OpenTabletDriver.Daemon.DriverDaemon.LoadUserSettings()
Jun 14 20:49:48 simon-thinkpad OpenTabletDriver.Daemon[1043545]:    at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__128_1(Object state)
Jun 14 20:49:48 simon-thinkpad OpenTabletDriver.Daemon[1043545]:    at System.Threading.QueueUserWorkItemCallbackDefaultContext.Execute()
Jun 14 20:49:48 simon-thinkpad OpenTabletDriver.Daemon[1043545]:    at System.Threading.ThreadPoolWorkQueue.Dispatch()
Jun 14 20:49:48 simon-thinkpad OpenTabletDriver.Daemon[1043545]:    at System.Threading.PortableThreadPool.WorkerThread.WorkerThreadStart()
Jun 14 20:49:48 simon-thinkpad OpenTabletDriver.Daemon[1043545]:    at System.Threading.Thread.StartCallback()
Jun 14 20:49:49 simon-thinkpad systemd[2649]: opentabletdriver.service: Main process exited, code=killed, status=6/ABRT
Jun 14 20:49:49 simon-thinkpad systemd[2649]: opentabletdriver.service: Failed with result 'signal'.
Jun 14 20:50:00 simon-thinkpad /usr/libexec/gdm-wayland-session[760680]: [2022-06-14 21:50:00.000] [error] : argument not found
Jun 14 20:50:03 simon-thinkpad firefox[1024525]: Unable to load sb_h_double_arrow from the cursor theme
Jun 14 20:50:18 simon-thinkpad /usr/libexec/gdm-wayland-session[760658]: 13:00:27.853 [ERROR] [wlr] [xwayland/xwm.c:1564] xcb error: op 18:0, code 3, sequence 11232, value 16777220
Jun 14 20:50:19 simon-thinkpad systemd[2649]: opentabletdriver.service: Scheduled restart job, restart counter is at 5.
Jun 14 20:50:19 simon-thinkpad systemd[2649]: Stopped OpenTabletDriver Daemon.
Jun 14 20:50:19 simon-thinkpad systemd[2649]: Started OpenTabletDriver Daemon.
```

jamesbt365, 2022-06-14:
does it get any further by manually starting the deamon (not by the service)?

SimonWoolf, 2022-06-14:
Yes, looks like if I just run `/usr/share/OpenTabletDriver/OpenTabletDriver.Daemon -c /usr/share/OpenTabletDriver/Configuration` in a terminal it starts absolutely fine, no crash.

jamesbt365, 2022-06-14:
Try one of the solutions here: OpenTabletDriver/OpenTabletDriver.Web/issues/43

SimonWoolf, 2022-06-14:
That worked, thanks! (specifically, execing `/usr/share/OpenTabletDriver/OpenTabletDriver.Daemon &` from sway_config (after setting up the env vars) worked, not invoking systemctl from sway_config).

jamesbt365, 2022-06-14:
Closing as services are made by the maintainers and  solution has been provided

closed 2022-06-14 by jamesbt365