OpenTabletDriver/OpenTabletDriver #2320: Memory leak of main UI on Linux (UX.Gtk)

MrApplejuice, 2022-06-21:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Steps to Reproduce

Basic steps:
- Leave the application open for a long time (8 hours+).
- Maybe: Connect and disconnect a graphics tablet a lot (I might have used a shoddy USB cable and the tablet might have disconnected and reconnected for hours on end)

System:
Xubuntu 20.04

```
paul@amd3900x:~$ uname -a
Linux amd3900x 5.4.0-120-generic #136-Ubuntu SMP Fri Jun 10 13:40:48 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
```

### Expected Behavior

Application uses a lot less than 30 GB of memory.

### Observed Behavior

When I woke up my computer from standby, I noticed that half of my system memory (64 GB) was occupied, most of it by OpenTableDriver.UX.Gtk.

Here the relevant `top` output:

~~~~~~~~~~~~~~~~
top - 11:03:24 up 3 days, 17:34,  2 users,  load average: 4,39, 3,17, 3,06
Tasks: 670 total,   2 running, 667 sleeping,   0 stopped,   1 zombie
%Cpu(s):  8,8 us,  0,0 sy,  0,0 ni, 91,2 id,  0,0 wa,  0,0 hi,  0,0 si,  0,0 st
MiB Mem :  64245,9 total,   6719,1 free,  42742,2 used,  14784,6 buff/cache
MiB Swap:      0,0 total,      0,0 free,      0,0 used.  20080,9 avail Mem 

    PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND                                                                                                  
 338458 paul      20   0   34,2g  29,8g  89712 S   0,0  47,5  36:23.76 /usr/share/OpenTabletDriver/OpenTabletDriver.UX.Gtk                                                      
 272972 paul      20   0 6340512   2,2g  44312 S   0,0   3,4   6:05.11 /usr/share/OpenTabletDriver/OpenTabletDriver.Daemon -c /usr/share/OpenTabletDriver/Configurations    
~~~~~~~~~~~~~~~~

This is the view that was open:

[image]

### Tablet Device

Huion S610

### Diagnostic Information

I got an error while saving the diagnostics:

[image]

The diagnostics-file ended up being 200 MB in size. So I zipped it and attached it that way.

[diag.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/8947446/diag.zip)

<details>
  <summary>Here a tail and head of that file</summary>

HEAD
```.json
{
  "App Version": "OpenTabletDriver v0.6.0.4",
  "Build Date": "05/04/2022",
  "Operating System": {
[... 878 lines cut ...]
      "Message": "Pen Bindings: OpenTabletDriver.Desktop.Binding.LinuxArtistMode.LinuxArtistModeButtonBinding, ",
      "StackTrace": null,
      "Level": 1,
      "Notification": false
    },
    {
      "Time": "2022-06-21T11:09:20.6381134+02:00",
      "Group": "Huion HS610",
      "Message": "Express Key Bindings: Key Binding: Subtract, Key Binding: Plus, , , Multi-Key Binding: Control+Y, Multi-Key Binding: Control+Z, , , , , , ",
      "StackTrace": null,
      "Level": 1,
      "Notification": false
    },
    {
      "Time": "2022-06-21T11:09:20.6381188+02:00",
      "Group": "Settings",
      "Message": "Driver is enabled.",
      "StackTrace": null,
      "Level": 1,
      "Notification": false
    }
  ]
}
```
</details>

MrApplejuice, 2022-06-21:
This _might_ be a duplicate of https://github.com/OpenTabletDriver/OpenTabletDriver/issues/2200, but I am not sure...

MrApplejuice, 2022-06-21:
Thanks for looking into this!

I was not sure about the large log-file being the only issue: 7.8 million times - ~100 bytes(?) would be an order of magnitude smaller at ~1 GB than 30 GB. So I would expect something else might be going on as well. However, I am really not super familiar with .NET or this project.

Maybe some very delayed garbage-collection that needs to be contained on systems with large memory or something could also be the reason? Admittedly, I am just guessing at this point :-)

jamesbt365, 2022-06-28:
This is under every platform and will be addressed by the undergoing rework on both the ui and daemon.

AkiSakurai, 2022-06-29:
> Seeing the same under OSX Big Sur 11.4 (20F71). OpenTabletDriver v0.5.3.3
> 
> [image]
> 
> I believe I had left it running for weeks. No filters/tools/interpolators. Daily disconnect/reconnect when using USB dock.
> 
> May I suggest a quick fix for this issue and possibly #2200 (if it's the same issue) would be to have a checkbox, disabled by default, to disable the diagnostics tab / logging? So unless you actually want all that diagnostics generated, it doesn't bother?

You are running old version of otd.

jamesbt365, 2022-09-18:
This needs to be tested again in master

MrApplejuice, 2022-10-20:
Hey - got around compiling `master` two days ago so. Unfortunately the memory leak is still there I think. Here the top-output::

~~~~~~~~~~~~~~~~~~~~
top - 21:44:20 up 25 days, 12:33,  5 users,  load average: 1,28, 1,56, 3,47
Tasks: 601 total,   1 running, 600 sleeping,   0 stopped,   0 zombie
%Cpu(s):  3,2 us,  0,6 sy,  0,0 ni, 96,0 id,  0,0 wa,  0,0 hi,  0,2 si,  0,0 st
MiB Mem :  64245,8 total,   6943,2 free,  21211,1 used,  36091,5 buff/cache
MiB Swap:      0,0 total,      0,0 free,      0,0 used.  34229,3 avail Mem 

    PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND                                                                                                                                                                
 278852 paul      20   0   10,3g   5,9g  79,5m S   0,0   9,3   6:18.25 /usr/share/OpenTabletDriver/OpenTabletDriver.UX.Gtk                                                                                                                    
3687494 paul      20   0   17,4g   4,1g  43,8m S   0,7   6,6 174:09.02 /home/paul/local/android-studio/jre/bin/java -classpath /home/paul/local/android-studio/lib/bootstrap.jar:/home/paul/local/android-studio/lib/util.jar:/home/paul/loc+ 
 278319 paul      20   0 5730,9m   1,6g  45,2m S   0,0   2,6   2:25.41 /usr/share/OpenTabletDriver/OpenTabletDriver.Daemon -c /usr/share/OpenTabletDriver/Configurations 
[...]
~~~~~~~~~~~~~~~~~~~~

It is not (yet) as severe as it used to be, but from the program's behavior, I thing it would have gotten to the point. What I observed was that after waking up my PC from sleep, the OpenTabletUX that I left open would detect my Huion-tablet, but then loose it again, detect it again and loose it again, and so on. Disconnecting and reconnecting the tablet solved the problem.

I extracted the logs again and it again is already 1.7 million lines long. :

<details>
  <summary>Here the head</summary>
~~~~~~~~~~
{
  "AppVersion": "OpenTabletDriver v0.7.0.0+89e2f525",
  "BuildDate": "",
  "OperatingSystem": {
    "Platform": 4,
    "ServicePack": "",
    "Version": "5.4.0.126",
    "VersionString": "Unix 5.4.0.126"
  },
  "EnvironmentVariables": {
    "USER": "paul",
    "DISPLAY": ":0",
    "PWD": "/home/paul",
    "PATH": "/home/paul/.cargo/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/home/paul/.dotnet/tools"
  },
  "Devices": [
    {
      "ProductID": 49243,
      "VendorID": 1133,
      "InputReportLength": 7,
      "OutputReportLength": 0,
      "FeatureReportLength": 0,
      "Manufacturer": "Logitech",
      "ProductName": "USB Optical Mouse",
      "FriendlyName": "USB Optical Mouse",
      "SerialNumber": "",
      "DevicePath": "/sys/devices/pci0000:00/0000:00:01.3/0000:02:00.0/usb1/1-4/1-4.2/1-4.2:1.0/0003:046D:C05B.0037/hidraw/hidraw0",
      "CanOpen": true
    },
    {
      "ProductID": 1031,
      "VendorID": 4176,
      "InputReportLength": 65,
      "OutputReportLength": 65,
      "FeatureReportLength": 0,
      "Manufacturer": "Yubico",
      "ProductName": "YubiKey OTP+FIDO+CCID",
      "FriendlyName": "YubiKey OTP+FIDO+CCID",
      "SerialNumber": "0019652419",
      "DevicePath": "/sys/devices/pci0000:00/0000:00:01.3/0000:02:00.0/usb1/1-4/1-4.3/1-4.3:1.1/0003:1050:0407.0053/hidraw/hidraw8",
      "CanOpen": true
    },
    {
      "ProductID": 5666,
      "VendorID": 4152,
      "InputReportLength": 6,
      "OutputReportLength": 0,
[... 106 lines cut ...]
      "ProductName": "Huion Tablet",
      "FriendlyName": "Huion Tablet",
      "SerialNumber": "",
      "DevicePath": "/sys/devices/pci0000:00/0000:00:01.3/0000:02:00.0/usb1/1-5/1-5:1.0/0003:256C:006D.0089/hidraw/hidraw13",
      "CanOpen": true
    }
  ],
  "ConsoleLog": [
    {
      "Time": "2022-10-20T21:34:55.8496701+02:00",
      "Group": "Detect",
      "Message": "Searching for tablets...",
      "StackTrace": null,
      "Level": 1,
      "Notification": false
    },
[...]
~~~~~~~~~~
</details>

And I attached big chunk of the tail (30k lines) to this post: [log-tail.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/9833497/log-tail.txt)

What strikes me as suspicious is that the detection logs at the end seem to search for new devices every 20 milliseconds, filling up the log in no time. Maybe that is another bug?

Anyway, that bug seemingly still fills up my memory, also of the daemon. Restarting the UI brought back its memory use (RSS) to a more manageable 160 MB, restarting the Daemon reduced its memory use to 100 MB.

jamesbt365, 2022-10-21:
Nope. None of those. The mass logging is not caused from any usb problems but a flaw in the sleep detection logic. That is fixed in #2236

MrApplejuice, 2022-10-22:
> @MrApplejuice Suggestion: first rule out hardware issues. [...]

Hey, thanks for the tip. Brown-outs were actually an issue in the past for the tablet when 
I tried to connect it to a (powered!) USB-Hub. As a mitigation, I use a USB 2.0 port of my PC directly now. Still, thank you a lot for helping out and debugging this.

> Nope. None of those. The mass logging is not caused from any usb problems but a flaw in the sleep detection logic. That is fixed in #2236

That sounds like it might be the issue for causing the big logging spike. Thanks for the PR, I will wait until it reaches `master` and then try again!