OpenTabletDriver/OpenTabletDriver #2321: Xp-Pen Star G640S Buttons Not Being Detected

ohitstom, 2022-06-22:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Steps to Reproduce

1. Open tablet driver
2. Open auxiliary tab
3. attempt to make binding
4. buttons on tablet not being detected

### Expected Behavior

To be able to bind my tablets buttons

### Observed Behavior

No button inputs being detected

### Tablet Device

Xp-pen g640s

### Diagnostic Information

[diagnostics.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/8958297/diagnostics.txt)

ohitstom, 2022-06-22:
it doesnt show up at all, pressing the buttons on the tablet does absolutely nothing in the software, there is nothing to save...

ohitstom, 2022-06-22:
I cannot create a binding, so no there isnt. When i attempt to create a binding and the window comes up asking for input, i press the buttons on my tablet and it does not detect them. Saving the binding leaves it blank, saving my profile and applying my profile also does nothing, since the binding was not able to be created.

X9VoiD, 2022-06-22:
Closing as user error, bindings are set by clicking on which tablet button you want to bind, *and then the button you want it to emit upon pressing the tablet button*, or with advanced binding editor you pick it through a series of drop downs.

closed 2022-06-22 by X9VoiD

X9VoiD, 2022-06-22:
On the exact same hardware as you

Go to Auxiliary Settings tab
[image]

A popup will show up

And then you press what you want it to emit, in my case I pressed Windows Key + Shift.
[image]

And here it is applied
[image]

ohitstom, 2022-06-22:
ohhh my bad i completely misunderstood, i assume assigning stuff would work backwards