OpenTabletDriver/OpenTabletDriver #2323: Support Request: Tablet not detected and no permissions for /dev/hidraw3 - 5

Quantenzitrone, 2022-06-24:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### What do you need help with?

OpenTabletDriver doesn't detect my tablet, probably because its access to /dev/hidraw is denied.

I have `hardware.opentabletdriver.enable = true;` in my config.

In the NixOS documentation it is stated, that this is supposed to enable OTD Udev rules, which is, as far as I know with my still little Linux knowledge, what handles access to /dev/ stuff.

So I have no idea what went wrong.

### What have you tried?

I tried searching how to get access to /dev/hidraw or rather how to get raw access to HID devices, but I found nothing helpful.

### Diagnostic Information

```
[Detect:Info]	No tablets were detected.
[Detect:Info]	Searching for tablets...
[DeviceUnauthorizedAccessException:Error]	Not permitted to open HID class device at /dev/hidraw4.
   at HidSharp.Platform.Linux.LinuxHidStream.DeviceHandleFromPath(String path, HidDevice hidDevice, oflag oflag)
   at HidSharp.Platform.Linux.LinuxHidDevice.TryParseReportDescriptor(ReportDescriptor& parser, Byte[]& reportDescriptor)
   at HidSharp.Platform.Linux.LinuxHidDevice.RequiresGetInfo()
   at HidSharp.Platform.Linux.LinuxHidDevice.GetMaxInputReportLength()
   at OpenTabletDriver.Devices.HidSharpBackend.HidSharpEndpoint.<>c.<get_InputReportLength>b__7_0(HidDevice d) in /build/source/OpenTabletDriver/Devices/HidSharpBackend/HidSharpEndpoint.cs:line 18
   at OpenTabletDriver.Extensions.TryGet[TSource,TValue](TSource source, Func`2 predicate, TValue& value) in /build/source/OpenTabletDriver/Extensions.cs:line 12
[DeviceUnauthorizedAccessException:Error]	Not permitted to open HID class device at /dev/hidraw5.
   at HidSharp.Platform.Linux.LinuxHidStream.DeviceHandleFromPath(String path, HidDevice hidDevice, oflag oflag)
   at HidSharp.Platform.Linux.LinuxHidDevice.TryParseReportDescriptor(ReportDescriptor& parser, Byte[]& reportDescriptor)
   at HidSharp.Platform.Linux.LinuxHidDevice.RequiresGetInfo()
   at HidSharp.Platform.Linux.LinuxHidDevice.GetMaxInputReportLength()
   at OpenTabletDriver.Devices.HidSharpBackend.HidSharpEndpoint.<>c.<get_InputReportLength>b__7_0(HidDevice d) in /build/source/OpenTabletDriver/Devices/HidSharpBackend/HidSharpEndpoint.cs:line 18
   at OpenTabletDriver.Extensions.TryGet[TSource,TValue](TSource source, Func`2 predicate, TValue& value) in /build/source/OpenTabletDriver/Extensions.cs:line 12
[DeviceUnauthorizedAccessException:Error]	Not permitted to open HID class device at /dev/hidraw3.
   at HidSharp.Platform.Linux.LinuxHidStream.DeviceHandleFromPath(String path, HidDevice hidDevice, oflag oflag)
   at HidSharp.Platform.Linux.LinuxHidDevice.TryParseReportDescriptor(ReportDescriptor& parser, Byte[]& reportDescriptor)
   at HidSharp.Platform.Linux.LinuxHidDevice.RequiresGetInfo()
   at HidSharp.Platform.Linux.LinuxHidDevice.GetMaxInputReportLength()
   at OpenTabletDriver.Devices.HidSharpBackend.HidSharpEndpoint.<>c.<get_InputReportLength>b__7_0(HidDevice d) in /build/source/OpenTabletDriver/Devices/HidSharpBackend/HidSharpEndpoint.cs:line 18
   at OpenTabletDriver.Extensions.TryGet[TSource,TValue](TSource source, Func`2 predicate, TValue& value) in /build/source/OpenTabletDriver/Extensions.cs:line 12
[DeviceUnauthorizedAccessException:Error]	Not permitted to open HID class device at /dev/hidraw4.
   at HidSharp.Platform.Linux.LinuxHidStream.DeviceHandleFromPath(String path, HidDevice hidDevice, oflag oflag)
   at HidSharp.Platform.Linux.LinuxHidDevice.TryParseReportDescriptor(ReportDescriptor& parser, Byte[]& reportDescriptor)
   at HidSharp.Platform.Linux.LinuxHidDevice.RequiresGetInfo()
   at HidSharp.Platform.Linux.LinuxHidDevice.GetMaxInputReportLength()
   at OpenTabletDriver.Devices.HidSharpBackend.HidSharpEndpoint.<>c.<get_InputReportLength>b__7_0(HidDevice d) in /build/source/OpenTabletDriver/Devices/HidSharpBackend/HidSharpEndpoint.cs:line 18
   at OpenTabletDriver.Extensions.TryGet[TSource,TValue](TSource source, Func`2 predicate, TValue& value) in /build/source/OpenTabletDriver/Extensions.cs:line 12
[DeviceUnauthorizedAccessException:Error]	Not permitted to open HID class device at /dev/hidraw5.
   at HidSharp.Platform.Linux.LinuxHidStream.DeviceHandleFromPath(String path, HidDevice hidDevice, oflag oflag)
   at HidSharp.Platform.Linux.LinuxHidDevice.TryParseReportDescriptor(ReportDescriptor& parser, Byte[]& reportDescriptor)
   at HidSharp.Platform.Linux.LinuxHidDevice.RequiresGetInfo()
   at HidSharp.Platform.Linux.LinuxHidDevice.GetMaxInputReportLength()
   at OpenTabletDriver.Devices.HidSharpBackend.HidSharpEndpoint.<>c.<get_InputReportLength>b__7_0(HidDevice d) in /build/source/OpenTabletDriver/Devices/HidSharpBackend/HidSharpEndpoint.cs:line 18
   at OpenTabletDriver.Extensions.TryGet[TSource,TValue](TSource source, Func`2 predicate, TValue& value) in /build/source/OpenTabletDriver/Extensions.cs:line 12
[DeviceUnauthorizedAccessException:Error]	Not permitted to open HID class device at /dev/hidraw3.
   at HidSharp.Platform.Linux.LinuxHidStream.DeviceHandleFromPath(String path, HidDevice hidDevice, oflag oflag)
   at HidSharp.Platform.Linux.LinuxHidDevice.TryParseReportDescriptor(ReportDescriptor& parser, Byte[]& reportDescriptor)
   at HidSharp.Platform.Linux.LinuxHidDevice.RequiresGetInfo()
   at HidSharp.Platform.Linux.LinuxHidDevice.GetMaxInputReportLength()
   at OpenTabletDriver.Devices.HidSharpBackend.HidSharpEndpoint.<>c.<get_InputReportLength>b__7_0(HidDevice d) in /build/source/OpenTabletDriver/Devices/HidSharpBackend/HidSharpEndpoint.cs:line 18
   at OpenTabletDriver.Extensions.TryGet[TSource,TValue](TSource source, Func`2 predicate, TValue& value) in /build/source/OpenTabletDriver/Extensions.cs:line 12
[Detect:Info]	No tablets were detected.
[Settings:Info]	Presets have been refreshed. Loaded 0 presets.
```

closed 2022-06-24 by jamesbt365

Quantenzitrone, 2022-06-24:
Thanks, but how do I grab udev rules?

jamesbt365, 2022-06-24:
Just go into one of action builds, go down to artifacts, grab one of the udev rules from there and then extract it and take the parts that is required. I'll link one for you. https://github.com/OpenTabletDriver/OpenTabletDriver/suites/6988785703/artifacts/273645421

ramirez7, 2022-06-29:
@jamesbt365 @Quantenzitrone I ran into this issue too - thanks for the help!