OpenTabletDriver/OpenTabletDriver #2324: Key (and Multi-Key) Bindings behave unintended when using a non-conventional Keyboard Layout

Quantenzitrone, 2022-06-24:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Steps to Reproduce

Have a XP-PEN Artist 12 2. Generation Pentablet (someone please try it with other tablets).
Install OpenTabletDriver or their proprietary Driver, as it has the same issue.
1. Select the [Bone-Keyboard-Layout](https://www.neo-layout.org/Layouts/bone/)(German) on (probably) any Linux-Distribution in my case NixOS.
  - on Gnome, you have to execute `gsettings set org.gnome.desktop.input-sources show-all-sources true` first
  - other instructions can be found on their website (in German) but since it is preinstalled on as far as I know every Linux distro, it should be easy.
2, Map any button with a letter key (because they are easy to observe)
  - üäöß (on cvb] in querty) won't work
3. Open any text editor or somewhere to type
4. press the key you set your letter to and observed the resulting letter

This will probably work with Neo2, Dvorak, Bépo or similar layouts too, but I haven't tested it yet.

### Expected Behavior

The keys (or multi-keys) that were set get typed correctly.

### Observed Behavior

### Here are the keys that get typed:
#### set keys as shown in the OTD app
JDUAXHLMW
CTIEOBNRSGQ
FVYZ,.K
#### typed keys when pressing said shortcut keys in order (uppercase for better readability)
NIHCVWBSZD
ÜXLUMÖYATOJ
EÄPF,.R
#### So:
J -> N
D -> I
U -> H
A -> C
etc.
### here are the bone and querty layouts for comparison
[image]
[image]

So you might be able to see what happens:
the typed key is the key, that is on the current keyboard layout in the position of the set key in the querty layout
For example:
- J is set
- key 7 in the second row (J in querty) gets typed
- resulting key: N (key 7 in the second row in Bone)

Also notice , and . get typed correctly as they are in the same position on both layouts.

### Tablet Device

XP-PEN Artist 12 2. Generation Pentablet

### Diagnostic Information

N/A output was too long for GitHub

jamesbt365, 2022-06-24:
duplicate of #1821

closed 2022-06-24 by jamesbt365

Quantenzitrone, 2022-06-24:
sorry i didnt find the other issue
i genuinely searched i promise