OpenTabletDriver/OpenTabletDriver #2325: [Windows] GUI unable to start despite daemon running

Interpause, 2022-06-25:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Steps to Reproduce

1. Download latest release v0.6.0.4
2. Delete `%LOCALAPPDATA%\OpenTabletDriver` if present
3. Extract to `E:\OpenTabletDriver.win-x64`
4. Go into properties of both `OpenTabletDriver.UX.Wpf.exe` and `OpenTabletDriver.Daemon.exe` to unblock them from Windows SmartScreen
5. Double-click `OpenTabletDriver.UX.Wpf.exe` and observe nothing opens & `%LOCALAPPDATA%\OpenTabletDriver` isn't created.
6. Double-click `OpenTabletDriver.UX.Wpf.exe` and observe normal startup behaviour, including correct detection of tablet model, `[Settings:Info] Drive is enabled.`, and the creation of `%LOCALAPPDATA%\OpenTabletDriver` (without `settings.json` however).
7. Double-click `OpenTabletDriver.UX.Wpf.exe` again but it still doesn't run.
8. Repeat 7. but inside `cmd` & observe nothing opens & there being no error message.

### Expected Behavior

UI opens and gives some sort of error message.

### Observed Behavior

Nothing opens.

### Tablet Device

XP-Pen Star G430S

### Diagnostic Information

Windows Version: 21H2 (22000.778) / KB5014668

jamesbt365, 2022-06-25:
This is actually to do with how stupid Windows can be sometimes putting directories on onedrive afaik, just move the files to a new directory (don't move the directory) and it should work. That or completely nuke onedrive.

Interpause, 2022-06-25:
Does not seem to be the cause in my case: I have double-confirmed Windows did not somehow move %LOCALAPPDATA% to my OneDrive, and tried running `OpenTabletDriver.UX.Wpf.exe` after moving both executables to my Downloads folder (which isn't in my OneDrive either).

jamesbt365, 2022-06-25:
Make an entirely new directory, move both the executables to it

Interpause, 2022-06-25:
[image]

Still does not work. Should I try WinDbg?

jamesbt365, 2022-06-25:
Weird how this isn't working, make sure OpenTabletDriver is dead (check task manager) then reopen the ux.

Interpause, 2022-06-25:
Nothing under Details sorted by Name. I'll try WinDbg tmrw to see if I can get any useful info for you besides my current Windows version.

Interpause, 2022-06-26:
This is what I got:

```
A JSON parsing exception occurred in [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App\6.0.6\Microsoft.WindowsDesktop.App.runtimeconfig.json], offset 0 (line 1, column 1): The document is empty.Invalid framework config.json [C:\Program Files\dotnet\shared\Microsoft.WindowsDesktop.App\6.0.6\Microsoft.WindowsDesktop.App.runtimeconfig.json]ModLoad: 00007ffb`c2860000 00007ffb`c2878000   C:\WINDOWS\SYSTEM32\kernel.appcore.dll
```

So could it be a broken Windows update to dotnet?

Interpause, 2022-07-01:
[image]

I have no idea how this happened. That said, it is likely an issue on my side then. I guess its time to ditch Microsoft.

closed 2022-07-01 by Interpause