OpenTabletDriver/OpenTabletDriver #2328: Huion Kamvas 16 position reports incorrectly for right hand side of screen

dajabe, 2022-06-26:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Steps to Reproduce

* Install Manjaro 5.17.15-1 with Xfce 4.16
* Install OTD
* Configure OTD to map to Kamvas 16 Pro screen directly
* Move pen across display to right hand 2cm

### Expected Behavior

* Pen works fine on rest of screen
* Expect to get same results on the right hand 2cm of tablet

### Observed Behavior

*Cursor teleports back to left hand side of screen for the right most 2cm
* Tablet debugger observes reported position x-axis returning to 0 unexpectedly

### Tablet Device

Huion Kamvas Pro 16

### Diagnostic Information

[diags.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/8986409/diags.txt)

## More info

I've looked through multiple issues so far trying to find anyone else with a similar issue and I'm unable to find it. I'm wondering if it has something to do with Xfce specifically. I'm happy to do further troubleshooting, like installing alternative GUI's in August. I do have a Windows system that I have yet to try loading the driver on to see if the issue occurs there as well. Let me know if that would be useful for your troubleshooting. 

Thank you for your time. This driver seems to be exactly what I neeeded to make this work for me and I can use most of the screen but just the 2cm down the right hand side seems to report the wrong position.

Tablet has been working fine on the windows install using the stock Huion drivers.

jamesbt365, 2022-06-26:
Misleading title, this is the Huion Kamvas Pro 16 btw, let me look into this.

jamesbt365, 2022-06-26:
Download this file, remove the .txt at the end and move it to /usr/share/OpenTabletDriver/Configurations, tablet > detect tablet then export diagnostic again.
[Kamvas Pro 16.json.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/8986430/Kamvas.Pro.16.json.txt)

jamesbt365, 2022-06-26:
Assuming from the heart emoji that this works correctly? I know from a previous record that it does or should

jamesbt365, 2022-06-26:
Going to need an answer for verification, also max x and y coordinates by moving the pen to the bottom right in the tablet debugger.

dajabe, 2022-06-26:
Sorry just meant that as a thank you and I'm going to try that.

I've since cloned the repo and built 0.6.0.4 and ran that for this test. I moved the configuration file (had to create the folder)
No improvement with any of these steps

Here is the new diags with the configuration file loaded as requested. Going to reload back to 0.6.0.3 and double check cause I don't understand all the plumbing here yet.
[KamvasPro16-diags.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/8986490/KamvasPro16-diags.txt)

Apologies on missing the Pro in the title.

jamesbt365, 2022-06-26:
What distro are you on and how did you install OpenTabletDriver?

dajabe, 2022-06-26:
Woah okay so 0.6.0.4 is not looking at that configuration file for some reason. Loading with 0.6.0.3 now correctly reports the position of the pen.

I'm running Manjaro, installed using AUR initially. After loading 0.6.0.4 I'm kinda tempted to get that going the new tweaks to the interface are nice.

I couldn't see this fix in the FAQs am I just blind?

jamesbt365, 2022-06-26:
Whatever, if its loading in 0.6.0.3 just fine then go into tablet > tablet debugger, move your pen to the bottom right of the tablet and get the max x and y position coordinates.

dajabe, 2022-06-26:
My apologies I wasn't clear. Yes I get to the full edge now with that config loading. And in the interest of clarity I wasn't asking for your help with 0.6.0.3 on a issue.

Coords requested:
[68840, 38720]

I'm guessing that this configuration file just adjusts for the correct MaxX and MaxY?

What I did ask was why I couldn't find this fairly simple fix in the FAQ.

dajabe, 2022-06-26:
Anyway thank for the workaround I really appreciate it. I can stop being frustrated by this and get some work done now.

jamesbt365, 2022-06-26:
Submitted a pull request for a parser change so will be fixed in the next version

PR #2329 (2022-07-17): Switch to GianoReportParser on Huion Kamvas Pro 16
fixes #2328
closed 2022-12-31 by X9VoiD