OpenTabletDriver/OpenTabletDriver #2330: Add support for XP-PEN Artist Pro 16

KainoaKanter, 2022-06-27:
### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Manufacturer

XP-PEN

### Product Name

Artist Pro 16

### Device Width

443.27

### Device Height

256.45

### Pressure Levels

8192

### Additional Information

Amazon link: https://www.amazon.com/dp/B09B7MNM2Q

### Diagnostic Information

```json
{
  "App Version": "OpenTabletDriver v0.6.0.4+325c28ca",
  "Build Date": "6/27/2022",
  "Operating System": {
    "Platform": 4,
    "ServicePack": "",
    "Version": "5.18.6.1",
    "VersionString": "Unix 5.18.6.1"
  },
  "Environment Variables": {
    "USER": "kainoa",
    "TEMP": null,
    "TMP": null,
    "TMPDIR": null,
    "DISPLAY": ":1",
    "WAYLAND_DISPLAY": "wayland-0",
    "PWD": "/usr/share/OpenTabletDriver",
    "PATH": "/usr/local/sbin:/usr/local/bin:/usr/bin:/opt/android-sdk/cmdline-tools/latest/bin:/opt/android-sdk/platform-tools:/opt/android-sdk/tools:/opt/android-sdk/tools/bin:/home/kainoa/.dotnet/tools:/usr/lib/jvm/default/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl"
  },
  "HID Devices": [
    {
      "DevicePath": "/sys/devices/pci0000:00/0000:00:08.1/0000:0d:00.3/usb5/5-3/5-3:1.2/0003:28BD:094B.0011/hidraw/hidraw6",
      "Manufacturer": "HANVON UGEE",
      "ProductName": "Artist Pro 16",
      "SerialNumber": "000000",
      "FriendlyName": "Artist Pro 16",
      "VendorID": 10429,
      "ProductID": 2379,
      "InputReportLength": 12,
      "OutputReportLength": 12,
      "FeatureReportLength": 0,
      "CanOpen": true
    },
    {
      "DevicePath": "/sys/devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:06:00.1/usb1/1-2/1-2:1.1/0003:6964:0080.0008/hidraw/hidraw9",
      "Manufacturer": "IDOBAO",
      "ProductName": "ID80",
      "SerialNumber": "0",
[... 1736 lines cut ...]
      "Group": "Plugin",
      "Message": "Loading plugin 'Tablet Debounce'",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2022-06-26T21:31:54.0585787-07:00",
      "Group": "Plugin",
      "Message": "Loading plugin 'AbstractQbit's Radial Follow Smoothing'",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2022-06-26T21:31:54.058585-07:00",
      "Group": "Settings",
      "Message": "Presets have been refreshed. Loaded 0 presets.",
      "StackTrace": null,
      "Level": 1,
      "Notification": false
    }
  ]
}
```

jamesbt365, 2022-06-27:
Remove .txt from the end of this filename and move it to /usr/share/OpenTabletDriver/Configurations then tablet > detect tablet
[Artist 16 Pro.json.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/8991753/Artist.16.Pro.json.txt)

If it detects we will move on to alignment and other parts of the supporting process.

KainoaKanter, 2022-06-27:
Ty, abt to test now. Since my main is unsuspended, my account @thatonecalculator will be responding from now on

jamesbt365, 2022-06-27:
Brilliant, go to tablet > tablet debugger, move your pen to the bottom right of the tablet and get the highest x and y coordinates that you can.

<sup><sup>also what theme is that its really nice

ThatOneCalculator, 2022-06-27:
> also what theme is that its really nice

Rose Pine, I'm one of the authors of the colorscheme and it's been ported to many different apps. The main site is https://rosepinetheme.com and the GTK theme is fan-made and available at https://github.com/Fausto-Korpsvart/Rose-Pine-GTK-Theme

jamesbt365, 2022-06-27:
see if you can get it any higher on any of the axis' do them separately if need be though sometimes tablets do have values like this

ThatOneCalculator, 2022-06-27:
From your config:

> ```
    "Digitizer": {
      "Width": 344.16,
      "Height": 193.59,
      "MaxX": 34416.0,
      "MaxY": 19359.0
    },
```

That doesn't seem right, as the width is 443.27 and the height is 256.45 as listed on Amazon

jamesbt365, 2022-06-27:
Yep, i repurposed another configuration before i could get the details from the max x and y because going off the product page can lead to inaccuracies because its either wrong because of rounding etc

jamesbt365, 2022-06-27:
That shouldn't get in the way at all, go into tablet > tablet debugger again, does pressing the pen buttons change them to true for each button respectfully? And what is the max pressure when pressing the pen down?

ThatOneCalculator, 2022-06-27:
- Max pressure: 8191 (not 8192)
- Buttons on pen works
- Eraser works
- Doesn't fully show, but pressing down side buttons/scrolling wheel does make text appear for a brief time.

jamesbt365, 2022-06-27:
After you do that it should have the correct active area and should map to the screen and tablet properly

jamesbt365, 2022-06-27:
Also check if the tablet's auxiliary buttons work when bound. No need to check the wheel as this will be unsupported until OpenTabletDriver has a proper way of handling wheels which will either be in 0.7 or onwards from that.

ThatOneCalculator, 2022-06-27:
> After you do that it should have the correct active area and should map to the screen and tablet properly

Works!!

ThatOneCalculator, 2022-06-27:
>

ThatOneCalculator, 2022-06-27:
> Also check if the tablet's auxiliary buttons work when bound.

Sure, give me a min

> No need to check the wheel as this will be unsupported until OpenTabletDriver has a proper way of handling wheels which will either be in 0.7 or onwards from that.

Any clue on 0.7 will be? I really like wheel for zooming

jamesbt365, 2022-06-27:
Also one last thing, by rotation do you mean tilt? Should be able to go to -60 and 60 on both axis' if it appears at all in the debugger.

>Any clue on 0.7 will be? I really like wheel for zooming

The main wait right now is the dependency injection rewrite, then everything under [the 0.6.1](https://github.com/OpenTabletDriver/OpenTabletDriver/milestone/23) and [the 0.7](https://github.com/OpenTabletDriver/OpenTabletDriver/milestone/19) milestones presumably.

ThatOneCalculator, 2022-06-27:
Yes, by rotation I meant tilt. One problem -- binding the pen buttons doesn't seem to work (despite being bound to middle and right click, they don't work)

[image]

jamesbt365, 2022-06-27:
Are you in artist mode?

ThatOneCalculator, 2022-06-27:
I tried all three artist mode bindings but they do nothing lol

jamesbt365, 2022-06-27:
Hmm thats weird, are you sure you set it up correctly?
http://opentabletdriver.net/Wiki/FAQ/Linux#artist-mode

jamesbt365, 2022-06-27:
I have also opened a pull request in #2332

jamesbt365, 2022-06-29:
Could you go to help > export diagnostic, save that file and send it here with the .txt extension?

ThatOneCalculator, 2022-06-29:
[diagnostic.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/9014218/diagnostic.txt)

jamesbt365, 2022-06-29:
For some reason the interface we were using vanished, try this one. [Artist 16 Pro.json.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/9014232/Artist.16.Pro.json.txt) (remember to remove the .txt)

ThatOneCalculator, 2022-06-29:
@jamesbt365 It now detects the tablet but I'm getting no mouse movement or input. The tablet debugger/visualizer gives me nothing either

jamesbt365, 2022-06-29:
Well the interface we were using which is definitely the correct one is gone for some reason and this one doesn't work because its not the right now which is weird because it shouldn't of vanished in the first place. Try changing the 

InputReportLength to 8, and replugging your tablet

ThatOneCalculator, 2022-06-29:
Sorry, I just realized I was panicking this whole time and using the pen for my OLD tablet. Using the proper pen works fine, very sorry for wasting your time :bow:

ThatOneCalculator, 2022-06-29:
Yep

jamesbt365, 2023-04-07:
Fixed in #2332

PR #2332 (2022-07-17): Add support for XP-Pen Artist 16 Pro
fixes #2330

Also changed the tablets.md for consistency (it was bugging me)
closed 2023-04-07 by jamesbt365