OpenTabletDriver/OpenTabletDriver #2333: Linux debugging help for "fail device streams" and "device in use" are not logged anymore

jamesbt365, 2022-06-30:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Steps to Reproduce

- Open OpenTabletDriver without udev rules or with the `hid_uclogic` kernel module loaded.
- invoke detection of a tablet
- The link for the respective wiki pages is not logged into the console.

### Expected Behavior

The logs here should be placed to assist the user in debugging
https://github.com/OpenTabletDriver/OpenTabletDriver/blob/master/OpenTabletDriver/Driver.cs#L96-L101
https://github.com/OpenTabletDriver/OpenTabletDriver/blob/master/OpenTabletDriver/Driver.cs#L106-L111

### Observed Behavior

These are never logged. This is regression from 0.5.3.3

### Tablet Device

N/A

### Diagnostic Information

N/A, diagnostics don't show the log but do show that the kernel module is loaded and that there is no permission for /dev/hidrawX

gonX, 2023-10-03:
We don't necessarily have to say "failed to open device streams" etc if that isn't actually what is being reported to us - if anything, the wiki FAQ should be updated to account for the new errors.

Some testing on `master` (6c4ffacb91997ef0) reveals (with Wacom PTH-851):

- With missing udev rules, and `wacom` module loaded:
```
[RootHub:Debug]	Initialized internal child hubs: HidSharpDeviceRootHub
[Detect:Warning]	Another tablet driver found: Wacom
[Detect:Error]	Detected input coming from Wacom driver
[Detect:Info]	Searching for tablets...
[Detect:Debug]	Attempting to match config 'Wacom PTH-851' to device 'Intuos5 touch L (/sys/devices/pci0000:00/0000:00:01.2/0000:01:00.0/0000:02:08.0/0000:04:00.1/usb1/1-4/1-4:1.1/0003:056A:0317.0009/hidraw/hidraw8)'
[Detect:Debug]	Attempting to match config 'Wacom PTH-851' to device 'Intuos5 touch L (/sys/devices/pci0000:00/0000:00:01.2/0000:01:00.0/0000:02:08.0/0000:04:00.1/usb1/1-4/1-4:1.0/0003:056A:0317.0008/hidraw/hidraw7)'
[Detect:Info]	No tablets were detected.
```

- With missing udev rules, `wacom` module NOT loaded:
```
[RootHub:Debug]	Initialized internal child hubs: HidSharpDeviceRootHub
[Detect:Info]	Searching for tablets...
[Detect:Debug]	Attempting to match config 'Wacom PTH-851' to device 'Intuos5 touch L (/sys/devices/pci0000:00/0000:00:01.2/0000:01:00.0/0000:02:08.0/0000:04:00.1/usb1/1-4/1-4:1.0/0003:056A:0317.0008/hidraw/hidraw7)'
[Detect:Debug]	Attempting to match config 'Wacom PTH-851' to device 'Intuos5 touch L (/sys/devices/pci0000:00/0000:00:01.2/0000:01:00.0/0000:02:08.0/0000:04:00.1/usb1/1-4/1-4:1.2/0003:056A:0317.000A/hidraw/hidraw9)'
[Detect:Debug]	Attempting to match config 'Wacom PTH-851' to device 'Intuos5 touch L (/sys/devices/pci0000:00/0000:00:01.2/0000:01:00.0/0000:02:08.0/0000:04:00.1/usb1/1-4/1-4:1.1/0003:056A:0317.0009/hidraw/hidraw8)'
[Detect:Info]	No tablets were detected.
```

Diagnostics report with module loaded:
```
./OpenTabletDriver.Console diagnostics | jq '.Devices[] | select(.Manufacturer == "Wacom Co.,Ltd.")'
{
  "VendorID": 1386,
  "ProductID": 791,
  "InputReportLength": -1,
  "OutputReportLength": -1,
  "FeatureReportLength": -1,
  "Manufacturer": "Wacom Co.,Ltd.",
  "ProductName": "Intuos5 touch L",
  "FriendlyName": "Intuos5 touch L",
  "SerialNumber": "",
  "DevicePath": "/sys/devices/pci0000:00/0000:00:01.2/0000:01:00.0/0000:02:08.0/0000:04:00.1/usb1/1-4/1-4:1.1/0003:056A:0317.0009/hidraw/hidraw8",
  "CanOpen": true
}
{
  "VendorID": 1386,
  "ProductID": 791,
  "InputReportLength": -1,
  "OutputReportLength": -1,
  "FeatureReportLength": -1,
  "Manufacturer": "Wacom Co.,Ltd.",
  "ProductName": "Intuos5 touch L",
  "FriendlyName": "Intuos5 touch L",
  "SerialNumber": "",
  "DevicePath": "/sys/devices/pci0000:00/0000:00:01.2/0000:01:00.0/0000:02:08.0/0000:04:00.1/usb1/1-4/1-4:1.0/0003:056A:0317.0008/hidraw/hidraw7",
  "CanOpen": true
}
```
(with `wacom` module unloaded it shows an additional matching endpoint)

Didn't test `hid_uclogic` because the result is likely the same.

Summed up, the real issue is that we should be reporting missing udev rules. This seems reliable to report if we check for `fooReportLength` == `-1`

jamesbt365, 2023-10-03:
I recall saying previously that `-1` report lengths doesn't necessarily mean that no udev rules exist, it generally just means *no permissions* and this also happens on 1 interface when the hid_uclogic module is using a tablet, regardless of if udev rules are present.